graspnet/graspnet-baseline #2: ModuleNotFoundError: No module named 'knn_pytorch'

cww97, 2021-01-03:
```sh
$ bash command_train.sh
Traceback (most recent call last):
  File "train.py", line 21, in <module>
    from graspnet import GraspNet, get_loss
  File "/home/weiwen/mnt0/mail2020/graspDL/graspnet-baseline/models/graspnet.py", line 22, in <module>
    from label_generation import process_grasp_labels, match_grasp_view_and_label, batch_viewpoint_params_to_matrix
  File "/home/weiwen/mnt0/mail2020/graspDL/graspnet-baseline/utils/label_generation.py", line 14, in <module>
    from knn_modules import knn
  File "/home/weiwen/mnt0/mail2020/graspDL/graspnet-baseline/knn/knn_modules.py", line 7, in <module>
    from knn_pytorch import knn_pytorch
ModuleNotFoundError: No module named 'knn_pytorch'
```

did I miss sth ?

chenxi-wang, 2021-01-04:
Hi, could you please provide your python version? In python 3.7 I cannot find this error.
Or have you install the knn module according to readme?

cww97, 2021-01-04:
sorry, I am blind

however, sth else get wrong

```bash
$ CUDA_VISIBLE_DEVICES=0 python train.py --camera realsense --log_dir logs/log_rs --batch_size 2 --dataset_root /home/weiwen/mnt0/mail2020/graspDL/datasets/graspNet-1b
Loading grasping labels...:  27%|████████████████████████████████████▎                                                                                                | 24/88 [01:35<02:40,  2.52s/it][1]    15812 killed     CUDA_VISIBLE_DEVICES=0 python train.py --camera realsense --log_dir   2

```

Fang-Haoshu, 2021-01-04:
Hi, can you monitor the cpu memory during loading? I guess it is killed by oom

cww97, 2021-01-04:
yes, u r right. let try this code on a huge machine after tomorrow, thank u very much

closed 2021-01-04 by cww97