graspnet/graspnet-baseline #3: meaning value of  meta.mat

gachiemchiep, 2021-01-05:
Hello graspnet team. 

Would you mind give me a quick explain about the meaning of each field inside your
 [meta.mat](https://github.com/graspnet/graspnet-baseline/blob/main/doc/example_data/meta.mat) file. 

```bash
{'__header__': b'MATLAB 5.0 MAT-file Platform: posix, Created on: Tue Mar 17 14:51:33 2020',
 '__version__': '1.0',
 '__globals__': [],
 'poses': array([[[-0.9756758 ,  0.20997444,  0.71809477, -0.10539012,
           0.28258908,  0.9570219 , -0.6063459 ,  0.6341392 ,
          -0.95235777],
         [-0.05635871,  0.7769375 , -0.6932231 , -0.9939491 ,
          -0.95306253, -0.1432863 , -0.11288764,  0.7661504 ,
          -0.22686528],
         [ 0.21185002, -0.59353083, -0.06149539, -0.0309534 ,
          -0.10869765, -0.25214702,  0.7871474 ,  0.10431238,
          -0.20383038],
         [ 0.1117    , -0.0714    , -0.0837    , -0.1092    ,
           0.0027    , -0.2043    , -0.0069    , -0.1581    ,
           0.0391    ]],
 
        [[ 0.21893798,  0.14432675, -0.64941216,  0.9167016 ,
           0.8435303 , -0.24861187,  0.7650126 , -0.66297054,
          -0.28011078],
         [-0.2993843 , -0.6250445 , -0.6992277 , -0.08504111,
           0.30085424,  0.04236081, -0.3529656 ,  0.6081769 ,
           0.915004  ],
         [ 0.9286739 , -0.76713043,  0.2989055 , -0.39041796,
          -0.44490823, -0.96767646,  0.53867525, -0.4365672 ,
           0.29035428],
         [ 0.0326    ,  0.0548    , -0.0858    ,  0.135     ,
          -0.0408    ,  0.0057    , -0.1437    , -0.0485    ,
           0.0349    ]],
 
        [[ 0.0110857 , -0.9669956 , -0.25020745,  0.38542327,
           0.45672753,  0.14933594,  0.21702617, -0.39791653,
           0.1206343 ],
         [ 0.95246667,  0.07541542, -0.17470661, -0.06952123,
           0.03403645,  0.9887743 ,  0.9288012 ,  0.20768833,
           0.33361623],
         [ 0.30444106, -0.24337623, -0.95229924,  0.9201172 ,
           0.8889553 ,  0.00491755,  0.30037972,  0.8936039 ,
          -0.9349586 ],
         [ 0.4197    ,  0.473     ,  0.4442    ,  0.5089    ,
           0.4402    ,  0.471     ,  0.3932    ,  0.4459    ,
           0.4803    ]]], dtype=float32),
 'cls_indexes': array([[ 1,  3,  6,  9, 23, 27, 52, 62, 39]]),
 'intrinsic_matrix': array([[631.54864502,   0.        , 638.43517329],
        [  0.        , 631.20751953, 366.49904066],
        [  0.        ,   0.        ,   1.        ]]),
 'factor_depth': array([[1000.]])}
```

Thanks

chenxi-wang, 2021-01-05:
Hi, the example data is copied from graspnet dataset.

Each image has a meta file containing object 6d poses, object ids, camera intrinsics and depth scale. It shares the same format with the meta file in YCB-Video dataset. Object ids and poses are labels used for training. For inference we only need camera intrinsics and depth scale. You can see https://graspnet.net/datasets.html for more details.

gachiemchiep, 2021-01-06:
@chenxi-wang 
Thank you for your explanation. 
I checked the [YCB_Video_toolbox](https://github.com/yuxng/YCB_Video_toolbox) but I still couldn't understand the meaning of the 'poses' variable. Would you mind give me a quick description of that. 

Thanks.

chenxi-wang, 2021-01-06:
“poses” contains object transformation matrix (translation & rotation) from object coordinates to camera coordinates.
poses[0:3, 0:3, i] and poses[0:3,3,i] denote the rotation matrix and the translation vector of the _i_-th object respectively.

gachiemchiep, 2021-01-06:
@chenxi-wang 
Thank you for your help

closed 2021-01-06 by gachiemchiep