graspnet/graspnet-baseline #5: I want to use your work, I want to build my own 3D data set, but what about the meta, grasp_label, dex_models file contents? Can you give me some information for reference? thank you

gzchenjiajun, 2021-01-12:
I want to use your work, I want to build my own 3D data set, but what about the meta, grasp_label, dex_models file contents? Can you give me some information for reference?
thank you

chenxi-wang, 2021-01-12:
Hi, ``meta`` contains object poses and camera intrinsics, sharing the same format with YCB-Video dataset (detailed at [Datasets](https://graspnet.net/datasets.html)). ``dex_models`` is generated from .obj & .sdf files using dexnet API, which is a cache for fast model loading and not necessary (see [Dexnet Model Cache](https://graspnetapi.readthedocs.io/en/latest/install.html?highlight=cache#dexnet-model-cache) for details.
The format of ``grasp_label`` is detailed at [Grasp Label Format](https://graspnetapi.readthedocs.io/en/latest/grasp_format.html). For grasp label generation, follow the process in [paper](https://openaccess.thecvf.com/content_CVPR_2020/papers/Fang_GraspNet-1Billion_A_Large-Scale_Benchmark_for_General_Object_Grasping_CVPR_2020_paper.pdf) to sample grasp candidates by single object grasp sampling & scene collision detection. Grasp quality annotation shares the same core function with [evaluation code](https://github.com/graspnet/graspnetAPI/blob/master/graspnetAPI/graspnet_eval.py).

gzchenjiajun, 2021-01-14:
> Hi, `meta` contains object poses and camera intrinsics, sharing the same format with YCB-Video dataset (detailed at [Datasets](https://graspnet.net/datasets.html)). `dex_models` is generated from [...]

Is there any more in-depth information on how to build my own 3D data set? Graspnet - Baseline for making datasets. Thank you

Fang-Haoshu, 2021-01-14:
You should refer to our paper in the readme. 
For building dataset, we don't know what is the  3D data set you are mentioning. You should clarify your question clearly.

closed 2021-01-14 by Fang-Haoshu

QiuJunning, 2023-10-29:
> Hi, `meta` contains object poses and camera intrinsics, sharing the same format with YCB-Video dataset (detailed at [Datasets](https://graspnet.net/datasets.html)). `dex_models` is generated from [...]

Hello, when I checked the evaluation code, I found that the .obj and .sdf of the target must be required. How can I generate the corresponding obj and sdf from the point cloud of an object? Can you share as many details as possible here? Looking forward to your reply, thank you!

chenxi-wang, 2023-10-30:
@QiuJunning You can use tools like Open3D or meshlab to generate .obj files from point cloud data. The .sdf files could be created using our grasp pose annotation tool. To apply for the tool, please fill in this form: https://forms.gle/7o4f18UHsw55nQ256

biosbeta, 2024-05-24:
> The .sdf files could be created using our grasp pose annotation tool.
请问获取抓取位姿标注工具？

biosbeta, 2024-05-24:
> @QiuJunning You can use tools like Open3D or meshlab to generate .obj files from point cloud data. The .sdf files could be created using our grasp pose annotation tool. To apply for the tool, please fill in this form: https://forms.gle/7o4f18UHsw55nQ256

请问获取抓取位姿标注工具？

QiuJunning, 2024-05-27:
@biosbeta Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.

biosbeta, 2024-05-27:
> @biosbeta Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.
你好，方便的话，可以私聊一下吗，想了解一下构建数据集的流程

biosbeta, 2024-05-27:
> @biosbeta Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.

> @biosbeta Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.

你好，方便的话，可以私聊一下吗，想了解一下构建数据集的流程

biosbeta, 2024-05-27:
> @biosbeta Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.
你好，方便的话，可以私聊一下吗，我想具体了解一下构建数据集的流程

QiuJunning, 2024-05-28:
@biosbeta 你好，欢迎邮箱联系junning.qiu@stu.xjtu.edu.cn

xiaozhangniaibu, 2024-07-03:
你好,我也已经填写表格，请邀请我使用标注工具 @chenxi-wang

lostloner, 2024-07-29:
Hello chenxi, thanks for your great work! It is really impressive. I have applied for the use of annotation tools, can you invite me?

tjc609, 2025-03-25:
> [@biosbeta](https://github.com/biosbeta) Hi! You can fill out the questionnaire in the link like Chenxi Wang said and apply for relevant tools from them.
你好，chenxi，谢谢你的出色工作！真的令人印象深刻。我已经申请使用注释工具，你能邀请我吗？@chenxi-wang

hbh1225, 2025-04-21:
> [@biosbeta](https://github.com/biosbeta) 你好，欢迎邮箱联系[junning.qiu@stu.xjtu.edu.cn](mailto:junning.qiu@stu.xjtu.edu.cn)

 I have applied for the use of annotation tools, can you invite me?

Fang-Haoshu, 2025-04-21:
Added

Maytheforcebewithu, 2025-05-15:
> Added

Hi, I've sent the application for the annotation tool, can you invite me? thank you!

Maytheforcebewithu, 2025-05-27:
> [@biosbeta](https://github.com/biosbeta) 你好，欢迎邮箱联系[junning.qiu@stu.xjtu.edu.cn](mailto:junning.qiu@stu.xjtu.edu.cn)

Hi, I have applied for the use of annotation tools, can you invite me?

Fang-Haoshu, 2025-05-28:
added @Maytheforcebewithu