graspnet/graspnet-baseline #6: Are there any other indicators of how accurate an object is to grasp? I would like to know how accurate the baseline is on common objects

gzchenjiajun, 2021-01-12:
Are there any other indicators of how accurate an object is to grasp? I would like to know how accurate the baseline is on common objects
[image]

chenxi-wang, 2021-01-12:
Hi, this baseline model is for grasp detection in cluttered scenes, and not designed for a specified object. You can use an object mask to see the result on a single object.

gzchenjiajun, 2021-01-14:
> Hi, this baseline model is for grasp detection in cluttered scenes, and not designed for a specified object. You can use an object mask to see the result on a single object.

OK, thank you.
I used it before
NVLABS / 6DOF - Graspnet Repo, but their data set is not scalable enough (we want to build our own data set), so we will start to prepare to use your work.
Thank you very much!

gzchenjiajun, 2021-01-14:
> OK, thank you.
> I used it before
> NVLABS / 6DOF - Graspnet Repo, but their data set is not scalable enough (we want to build our own data set), so we will start to prepare to use your work.
> Thank you [...]

For example, in meta data, I download a dataset and look at meta data, but I don't know how to construct poses, like in meta.
By the way, is there any recommended software for building these data sets? (2D detection tasks like rectLabel are usually performed with software like rectLabel)
thank you

gzchenjiajun, 2021-01-14:
> For example, in meta data, I download a dataset and look at meta data, but I don't know how to construct poses, like in meta.
> By the way, is there any recommended software for building these data [...]

I did a little bit of work, I had a little bit of understanding, that it poses maybe point clouds of concrete objects, right? But I see shape is 3*4*9/3*4*10/...
I'm not sure how to construct the data
thank you

Fang-Haoshu, 2021-01-14:
You should refer to our paper.
These are common concepts in 3D geometry like object pose. You should get these knowledge on internet beforehand.
Thanks

closed 2021-01-14 by Fang-Haoshu

gzchenjiajun, 2021-01-14:
> concepts

ok