graspnet/graspnet-baseline #7: What do you mean by grasp_height and grasp_depth?

gzchenjiajun, 2021-01-15:
What means by grasp_height and grasp_width?

I read the source code models/graspnet.py:79,butI don't understand what grasp_height and grasp_depth mean.

I understand other parameters. For example, grasp_width is the gripper's gripping width.

thanks

chenxi-wang, 2021-01-15:
Hi, you can find the definition in the figure at [Grasp Label Format](https://graspnetapi.readthedocs.io/en/latest/grasp_format.html#raw-label-format).

gzchenjiajun, 2021-01-15:
[image]
Is it width, height, depth of the object?

chenxi-wang, 2021-01-15:
No. They are all about grippers. The box stands for gripper coordinates.

gzchenjiajun, 2021-01-15:
> No. They are all about grippers. The box stands for gripper coordinates.

I seem to understand.
These parameters score, width, height, depth, rotation_matrix, translation are all based on GRASP.
"Width" grasp width "height" depth "grasp height
But to some extent, width,height and depth are not used in my arm, I just need the posture (because my arm [XARM] is fixed, not adjustable in width,height and depth).
Do I understand you correctly?

chenxi-wang, 2021-01-15:
Yes, you are right.

gzchenjiajun, 2021-01-15:
> Yes, you are right.

'Points' records the grasp center point Coordinates in model frame.

I have read the understanding of Points, and I feel that if I want to make the manipulator run, I still need a center of mass coordinate of the object, right? (This parameter is not provided in the source code.)
I have to get it myself. Do I understand that right?

gzchenjiajun, 2021-01-15:
> 'Points' records the grasp center point Coordinates in model frame.
>
> I have read the understanding of Points, and I feel that if I want to make the manipulator run, I still need a center of mass [...]

Because the current network output values are all based on GRASP Angle data, but lack the information of objects (especially the center of mass of objects)

chenxi-wang, 2021-01-15:
I can't get your idea. Grasp poses depend on local geometry features, and center of mass cannot be obtained from visual input. It is a model-free algorithm, with no specified objects.

gzchenjiajun, 2021-01-15:
> I can't get your idea. Grasp poses depend on local geometry features, and center of mass cannot be obtained from visual input. It is a model-free algorithm, with no specified objects.

I would like to ask, what coordinate system is the output value of grasp translation (XYZ) based on? (I guess based on the center of the image (depth/color)?)

I thought it was based on the center of mass, could I be wrong?

Fang-Haoshu, 2021-01-15:
based on camera coordinate.
Please refer to our paper

closed 2021-01-15 by Fang-Haoshu

gzchenjiajun, 2021-01-15:
Thank you very much for replying to me tonight!
Sorry, I will continue to refer to your article.

As a newcomer to this direction, I am not familiar with a lot of knowledge