graspnet/graspnet-baseline #8: Segmentation Fault with GPU and not support CPU mode

pingguomaggie, 2021-01-16:
Hi, thanks for sharing the code and it looks very interesting. I am try to run the code following the readme. But when I use GPU mode it runs into segmentation fault "net.to(device)". I tested both train.py and test.py and both have the same error. When I force device="CPU", it also fails. Do you know how to fix this? The detailed errors are follows:

[image]

[image]

My environment is:
     Ubuntu 18.04 + python 3.6 + cuda 10.1 + pytorch 1.6 

Thanks so much for your help!

chenxi-wang, 2021-01-16:
pointnet2 operators do not support CPU mode.
For GPU mode, could you provide more information during execution? For example, set CUDA_LAUNCHING_BLOCK=1 before running the program. You can also import and convert net without other code to see if this error is caused by model device conversion.

pingguomaggie, 2021-01-16:
Hi chenxi, thanks so much for your quick reply. I fix this problem! The GPU segmentation fault is due to the wrong link of pointnet and knn with cuda. There was two versions of cuda in my machine. I uninstalled one of them. But do not know why the pointnet and knn linked to the removed cuda. The problem is fix now by using the correct cuda link. Thanks again for your help! We can close this issue now!

closed 2021-01-16 by pingguomaggie