graspnet/graspnet-baseline #9: In the last step, is it necessary to artificially refine the gt grasp data set for generation?

gzchenjiajun, 2021-01-22:
“Loss Function For each candidate point, we assign it a binary label indicating whether it is graspable or not. First, points which are not on the objects are assigned negative labels. Next, for points on the objects, we found those who have at least one graspable ground-truth within 5mm radius neighbor area. Their graspable scores are assigned as 1. Finally, points on the objects but cannot ﬁnd reference ground-truth grasps are ignored, which do not contribute to the training objective.”

I didn't understand  this passage
are gt datasets generated directly from network results?

gzchenjiajun, 2021-01-22:
I only see gripper checks based on depth and collision, but is this really reliable?

gzchenjiajun, 2021-01-22:
sentence : "The result is computed based on physical rules, which is robust."
In the end, you don't really need to check, but you can directly determine the label results based on the program results, right?

gzchenjiajun, 2021-01-22:
sentence : "for each scene. According to statistics, the ratio of positive and negative labels in our dataset is around 1:2."
Sorry, I really don't understand very well

chenxi-wang, 2021-01-24:
Hi, grasp qualities are annotated using analytic method, but not only physical rules. Refer to grasp evaluation in [graspnetAPI](https://github.com/graspnet/graspnetAPI) for details.
Your question is about the dataset, while this repo is about the baseline model, the post-processing using depth & collision is just a part of grasp selection.

closed 2021-01-24 by chenxi-wang

gzchenjiajun, 2021-01-25:
In my understanding, first generate gripper from pointnegpd and then go to the dexnet of grasp-evaluation to refine it, right?

chenxi-wang, 2021-01-26:
Grippers are generated by dense sampling, which is not the same as pointnetgpd. Annotation part share a similar idea.

gzchenjiajun, 2021-01-27:
> Grippers are generated by dense sampling, which is not the same as pointnetgpd. Annotation part share a similar idea.

Sorry, I want to ask what does "dense sampling" mean specifically? Which module outputs the gripper dense sampling? (I understand that gripper dense sampling requires infer from the basic network, and then the evaluation can be carried out in the next step)

gzchenjiajun, 2021-01-28:
> Sorry, I want to ask what does "dense sampling" mean specifically? Which module outputs the gripper dense sampling? (I understand that gripper dense sampling requires infer from the basic network, [...]
"For each image, we densely annotate 6-DoF grasp poses by analytic computation of force closure [29]."
Is gripper generation based on [29]? [29] Do you have the relevant information?
I don't see it in the Graspnet Repo

gzchenjiajun, 2021-01-28:
> Grippers are generated by dense sampling, which is not the same as pointnetgpd. Annotation part share a similar idea.
>
>
> Sorry, I want to ask what does "dense sampling" mean specifically? Which [...]
“For each image, we densely annotate 6-DoF grasp poses by analytic computation of force closure [29]."
Is gripper generation based on [29]? [29] Do you have the relevant information?
I don't see it in the Graspnet Repo

gzchenjiajun, 2021-01-28:
"For each image, we densely annotate 6-DoF grasp poses by analytic computation of force closure [29]."
Is gripper generation based on [29]? [29] Do you have the relevant information?
I don't see it in the Graspnet Repo

chenxi-wang, 2021-01-28:
You may ask about the gt grasp sampling, and it is not a part in the network. Please refer "Grasp Pose Annotation" in our paper for more details.

gzchenjiajun, 2021-01-28:
Yes,I  ask about the gt grasp sampling....
I refer this
"Figure 3. Grasp pose annotation pipeline. The grasp point is ﬁrstly

sampled from point cloud. Then the grasp view, the in-plane rotation and the gripper depth are sampled and evaluated. Finally, the grasps are projected on the scene using the 6D pose of each object. Collision detection is also conducted to avoid the collision between grasps and background or other object."

I don't understand the process here. Is there any code reference in the repo?

m current understanding is that gripper sampling can be generated with Repo such as PointGDP/Graspnet-baseline, and then evaluated with DexNet. Is my understanding correct?

chenxi-wang, 2021-01-28:
Sorry, we do not have a plan to release the code of the whole annotation process yet, but core functions could be found in graspnetAPI.

> my current understanding is that gripper sampling can be generated with Repo such as PointGDP/Graspnet-baseline, and then evaluated with DexNet. Is my understanding correct?

Yes, you are right.

gzchenjiajun, 2021-01-28:
OK, thank you very much！ @chenxi-wang