graspnet/graspnet-baseline #14: Consuming too much memory when training and testing

mahaoxiang822, 2021-03-08:
When training graspnet-baseline in my own enviroment, I find that it will cosuming too much memory（about 60G）. GPU memory is about 10G when batch size=2. It's hard for me to train it on some low memory server. I'm not sure if this is a bug and if not, is there any solution to reduce it?

chenxi-wang, 2021-03-22:
Hi, training the total dataset requires a machine with big memory. To reduce memory consumption, you can try the following tips:
Choose part of the scenes containing only a subset of all models.
Downsample grasp labels (grasp points, grasp views, depths, angles).
Ignore some models in data loading, model training, label generation and loss computation.

Fang-Haoshu, 2021-03-22:
You may need a machine with over 128GB CPU memory

closed 2021-03-22 by Fang-Haoshu

LXZ1998-666, 2021-07-09:
Can you tell me the specific computer configuration when you train the code? Such as CPU and GPU, model and size

LXZ1998-666, 2021-07-09:
> You may need a machine with over 128GB CPU memory

Can you tell me the specific computer configuration when you train the code? Such as CPU and GPU, model and size