graspnet/graspnet-baseline #15: objectness_score

trungpham2606, 2021-03-19:
@Fang-Haoshu @chenxi-wang 
Thank for sharing great work !

I have 2 questions about your work :
  1. We dont have to normalize the point cloud to zero mean before feeding through the network ? 
  2. After feeding the input point cloud through the basemodel + approach model, the objectness here is more about the point is object or not rather than graspable or not as stated in paper. Did I misunderstand something here ?

Hope to hear your response soon.

chenxi-wang, 2021-03-22:
Hi, for the second question, we treat "graspable" points as the ones on objects.

For the first question, we do not shift point clouds here, but I believe centralizing the points in the workspace brings more stable results. 3D predictions are sensitive to point coordinates, so cloud center should be aligned with the workspace of your gripper. Zero mean, workspace cropping and random shifting(in data augmentation) are good ideas to improve model performance in real experiments.

trungpham2606, 2021-03-22:
@chenxi-wang thank you.

closed 2021-03-22 by chenxi-wang