graspnet/graspnet-baseline #16: run demo on windows

Twilight89, 2021-04-06:
Hi~
I want to run the forward demo on windows, but when I run "python setup.py install" in knn file, occured error:
vision.obj : error LNK2001: 无法解析的外部符号 "struct THCState * state" (?state@@3PEAUTHCState@@EA)
build\lib.win-amd64-3.6\knn_pytorch\knn_pytorch.cp36-win_amd64.pyd : fatal error LNK1120: 2 个无法解析的外部命令
error: command 'P:\\VisualStudio2019\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120
I don't know why, I run successfully on ubuntu, but I need to run on windows, can you help me?
Hope for your reply, thanks ^^

Fang-Haoshu, 2021-04-06:
Hi, for the knn module, you can refer to its original implementation for help. I guess it is not designed for windows.

closed 2021-04-06 by Fang-Haoshu

baixiaomihu, 2024-04-16:
Hi, I also have the same problem. But after changing this line of code according to your method. The knn compilation still reports the same error. How did you solve it? I look forward to your reply. @zingaltern @Twilight89

zingaltern, 2024-04-16:
the problem is address by this command, but other problem need change python3.7 to python 3.8. By the way, I successfully installed in wondows.