graspnet/graspnet-baseline #18: Is GraspNet more suitable for top-down view?

hetolin, 2021-07-05:
Hi, thanks for sharing your work. It really facilitates robotic grasping. I am using your code recently in the gazebo. I found that the generated grasp points are `on the edge of the table` or `inserts into the body of the can` as in the figure below. The elevation of the camera is set to 45 degrees in my case. Could I ask if the graspnet is more suitable for top-down view( nearly 90 degrees to be perpendicular to the table) or top-down grasping?

[image]

chenxi-wang, 2021-07-05:
Hi, graspnet is not limited to top-down view (although it depends on the training data), but point clouds in **camera coordinate frame** are recommended.
For grasps on the table, I think it is due to the wrong objectness mask used in [post-processing](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/models/graspnet.py#L77), usually caused by incorrect coordinate frames. Since the workspace is known, there are also other solutions including: 1) directly filtering the grasps on the table using a mask; 2) only keeping the foreground objects in point cloud pre-processing.
The grasp inserting into the body is an unavoidable problem for single-view point cloud. But in the figure, this type of grasps are in low confidence and easy to be filtered out.

hetolin, 2021-07-05:
Thanks for your prompt reply. In my testing case, I directly using the depth and RGB image subscribed from the kinect camera topi in `Function get_and_process_data(data_dir)` in `demo.py`. Therefore, the back-projected point cloud should be in the camera coordinate frame. In this case, could I only use the object mask to filter the background points to get a more reasonable grasps?

chenxi-wang, 2021-07-05:
Sure, you can try adding a mask on input cloud or output grasps.

hetolin, 2021-07-05:
Thanks. I will give it a try

closed 2021-07-05 by hetolin