graspnet/graspnet-baseline #19: How to use the grasp label "rotate matrix" in Real robot implement.

luyh20, 2021-07-09:
I want to use this model in real robot experiment. And I want to use the rotate matrix of grasp to get the TCP value——"rx, ry, rz". But I can't get a right final position of gripper when multiply TCP rotate matrix with grasp rotate matrix to get "rx, ry, rz", the TCP rotate matrix is from foundation support to TCP. The gripper direction didn't match with the result in open3d visualization. Could you give me some details about using the grasp rotate matrix? Thanks.

closed 2021-07-09 by luyh20

luyh20, 2021-10-18:
Maybe it's a setup problem that there is a Unit rotation matrix between open3d visualization and the real world. Try to find it until the gripper direction matches with the result in open3d visualization. In the next, relate this rotation matrix to the rotation matrix label.

luyh20, 2021-10-18:
> 请问，您是如何解决的呢？

Maybe it's a setup problem that there is a Unit rotation matrix between open3d visualization and the real world. Try to find it until the gripper direction matches with the result in open3d visualization. In the next, relate this rotation matrix to the rotation matrix label.

JackeyLa5, 2023-04-05:
The same problem is bothering me. I would like to ask you how you found this transformation matrix and what operations you did to make the real robot work normally? Thanks.

JackeyLa5, 2023-04-06:
Thank you. I have found the answer.

suoyike1, 2024-11-02:
hi @luyh20 @JackeyLa5 @chenxi-wang   I come across this problem recently, may I ask you guys how to solve this problem?
I get the grasp pose from Grasp().rotation_matrix and translation, but when i translate them to arm base frame, it is always wrong.
Is there any tanslation matrix that i forget? I will very appreiate it! Thanks you guys very much!