graspnet/graspnet-baseline #20: 训练终止

LXZ1998-666, 2021-07-20:
您好，我在进行graspnet模型训练大约５小时后，第一个epoch还没有结束时，程序突然变的异常缓慢，batch: 12800 。我的服务器环境为：RTX3090  CUDA11.1  Pytorch1.7  python3.6，CPU内存128GB，batch_size = 2,epoch = 18，加载了所有的数据，包括１００个测试场景和３０个见过的物体，消耗CPU内存大概７０G。
这是显卡占用情况，显存占用较高，但是功率很低：
[image]
这是截止到当时的训练结果：
[image]

请问您在训练的时候遇到过这个问题吗？最后的损失在0.6左右，请问您的模型训练多次时间，总损失为多少？感谢您的回复

LXZ1998-666, 2021-07-20:
看起来像是程序终止了训练，但是又没有退出。

chenxi-wang, 2021-07-20:
Hi, you can check the log to see whether it is caused by the crush of dataloader.
The overall loss should be ~0.6 after the 1st epoch.

chenxi-wang, 2021-07-21:
> 感谢您的回复，根据您的建议，调查后发现可能是 dataloader 多线程处理数据的问题，您的代码默认设置：num_workers=4，您在训练过程中，是否修改过 worker 的数量？

I didn't change num_workers in baseline training. You can try to reduce it.

LXZ1998-666, 2021-07-22:
谢谢您的回答，对我很有帮助。我尝试了两个办法，首先是num_works设置为0，只使用主线程加载，这能正常训练，但数据加载非常慢，无法发挥显卡的优势。第二个办法是在训练数据和评估数据转换的时候设置休眠，time.sleep(1)，这也可以正常训练。
除此之外，想咨询您另一个问题，您在训练的时候，显卡利用率高吗？我的num_works设置为8，甚至是16，显卡利用率只能到60%，我怀疑是CPU处理数据的能力不足，导致GPU大部分时间一直在等待数据。您遇到过这种问题吗？

chenxi-wang, 2021-07-22:
Yes, data processing is a bottleneck in training. The long processing time is mostly caused by (1) [computing workspace masks](https://github.com/graspnet/graspnet-baseline/blob/main/dataset/graspnet_dataset.py#L176) and (2) [removing invisible grasp points](https://github.com/graspnet/graspnet-baseline/blob/main/dataset/graspnet_dataset.py#L216). For (1), you can dump all the masks beforehand. For (2), you can disable this function by changing the flag "remove_invisible" in dataset (which will decrease label precision).
If you use RTX 3090, you can simply try increasing training batch size.

LXZ1998-666, 2021-07-22:
> Yes, data processing is a bottleneck in training. The long processing time is mostly caused by (1) [computing workspace [...]

感谢您的回答，对我非常有帮助，后期会尝试优化。现在可以关闭这个话题了。

closed 2021-07-22 by LXZ1998-666