graspnet/graspnet-baseline #22: load pretrain weight error

DLYL2019, 2021-07-23:
１．运行test.py时，torch.load()报错：
Traceback (most recent call last):
  File "test.py", line 60, in <module>
    checkpoint = torch.load(cfgs.checkpoint_path)
  File "/home/xup/anaconda3/envs/pytorch11/lib/python3.6/site-packages/torch/serialization.py", line 387, in load
    return _load(f, map_location, pickle_module, **pickle_load_args)
  File "/home/xup/anaconda3/envs/pytorch11/lib/python3.6/site-packages/torch/serialization.py", line 560, in _load
    raise RuntimeError("{} is a zip archive (did you mean to use torch.jit.load()?)".format(f.name))
RuntimeError: logs/log_rs/checkpoint.tar is a zip archive (did you mean to use torch.jit.load()?)
２．把checkpoint.tar解压后，把模型载入路径改为．／checkpoint_rs／archive／data.pkl，则报错：
Traceback (most recent call last):
  File ".../graspnet-baseline/test.py", line 60, in <module>
    checkpoint = torch.load(cfgs.checkpoint_path)
  File "/home/xup/anaconda3/envs/pytorch11/lib/python3.6/site-packages/torch/serialization.py", line 387, in load
    return _load(f, map_location, pickle_module, **pickle_load_args)
  File "/home/xup/anaconda3/envs/pytorch11/lib/python3.6/site-packages/torch/serialization.py", line 564, in _load
    magic_number = pickle_module.load(f, **pickle_load_args)
_pickle.UnpicklingError: A load persistent id instruction was encountered,
but no persistent_load function was specified.
通过网上查询，说该报错应该是torch版本问题，需要训练和预测时版本一致，但是我测了pytorch1.1/1.6.0/1.8.0都报同样的错误。
请问该预训练权重是pytorch什么版本报错的？或者知道是其他什么原因导致吗？谢谢

chenxi-wang, 2021-07-23:
I have not seen this error in training (for graspnet or other models). And "checkpoint.tar" should be loaded directly. PyTorch version used in this repo is 1.6.

closed 2021-07-29 by DLYL2019

DLYL2019, 2021-07-29:
This is indeed caused by pytorch version issues. I used pytorch1.6 version to directly load the "checkpoint.tar" and solved it successfully. Thank you for your reply .

mao8-explorer, 2023-11-10:
torch.load('weight/checkpoint-kn.tar') solve problem ...