graspnet/graspnet-baseline #23: Grasp Score greater than 1

FrancescoRosa3, 2021-08-04:
Hi everyone.
I am using your work in order to build a vision-based grasping system for my master thesis.
By running the demo.py and by observing the grasp score, there are grasps that have a score greater than 1, but from your paper, I would have expected a max grasp score of 1. 
So, why are there grasps scores greater than 1?
Thanks.

chenxi-wang, 2021-08-04:
Hi, original baseline implementation compute gt scores by s = 1.1 - u, with u in [0.0, 1.0].
In current implementation, we use s = 0 when u = 0 and s = -ln(u) when u is in [0.1, 1.0].
With this definition, gt scores are ranging from 0 to ln(10).
The definition can be found [here](https://github.com/graspnet/graspnet-baseline/blob/main/utils/label_generation.py#L115).

FrancescoRosa3, 2021-08-04:
Thanks for your quick response.
It is clear, so the grasp score ranges from 0 to ln(1.0), is it correct?
I have another question, your work seems to be "gripper agnostic", but from the grasp definition, I understand the following:
-  width, is the gripper opening,
-  height, is the height of the gripper tip,
- depth, should be the distance from the origin of the gripper coordinate frame and the first point of the object to be grasped.
is it correct?
Then there is an offset of 2 cm, that should define the distance from the origin of the gripper coordinate and the palm of the gripper, but if I have a gripper that does not have this offset, (it may be lower or greater), do I have the possibility to set this parameter?

chenxi-wang, 2021-08-05:
Grasp score ranges from 0 to **ln(10)** (not ln(1.0)).
Depth is the distance from the origin of gripper coordinate frame to the gripper tip, and contact points usually lie between them. Detailed definition can be found in [graspnetAPI](https://graspnetapi.readthedocs.io/en/latest/grasp_format.html#d-grasp).

The height and 2cm offset are used in collision detection and graspnet evaluation, and can be modified according to your requirements in the real experiments. The coordinates of gripper tip computed from modified gripper paramters and original graspnet format needs to be aligned.

FrancescoRosa3, 2021-08-06:
I am sorry for the typo.
Probably I am missing something but based on the comment:
_"Depth is the distance from the origin of gripper coordinate frame to the gripper tip, and contact points usually lie between them"_, this depth should be constant, because the origin of the gripper coordinate is fixed as well as its distance from the gripper tip, but after running the demo I have not observed this behavior. What am I missing?
Moreover, what do you mean with _"modified gripper paramters and original graspnet format needs to be aligned"_?
Thank you for your kind.

chenxi-wang, 2021-08-10:
In the output parameters, coordinates of grasp points are not equivalent to the gripper bottom. The position of gripper tip is computed by grasp point and depth (not a constant), and the position of the bottom is computed using tip coordinate and the fixed gripper length (a variable you can set by yourself).

FrancescoRosa3, 2021-08-15:
Thank you for all your clarifications.
After reading very carefully both the paper and the code, and thought about your comment, I have the following observations to do:

1. The **translation** field of a Gripper is the coordinate of the **grasp point** in the image coordinate frame, isn't it?
2. The offset of 2 cm should be equal to `hmin = -0.02`.
3. The **depth** can assume 4 values equal to `hmax_list=[0.01,0.02,0.03,0.04]`, but I do not understand why this value should be _"the distance from the origin of gripper coordinate frame to the gripper tip"_. It seems to me, that this value is used only [here](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/models/modules.py#L80), and then we take the height that produces the grasp with the highest score [here](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/models/graspnet.py#L103). Moreover, there is no _"gripper coordinate frame"_, what am I missing about the depth? What is its physical meaning?
4. If I want to modify the code in order to take into account my gripper geometry, I should modify: 
  
- [GRASP_MAX_WIDTH](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/utils/loss_utils.py#L8), in order to take into account my gripper max opening.
- [finger_width and finger_lenght](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/utils/collision_detector.py#L29), for the collision detection.
- [grasp_height](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/models/graspnet.py#L131), for the collision detection.
- [hmin and hmax_list](https://github.com/graspnet/graspnet-baseline/blob/df67956a1fecadf5b47ffe141103edfb8deec543/demo.py#L38).
- I should modify the corresponding variables [here](https://github.com/graspnet/graspnetAPI/blob/58d4aba855099b0f5f86fcca987e26baf1809179/graspnetAPI/utils/utils.py#L453), for a correct plot of the gripper. 

Thanks for your kind.

chenxi-wang, 2021-08-25:
> The translation field of a Gripper is the coordinate of the grasp point in the image coordinate frame, isn't it?

It should be **camera coordinate frame**, which is transformed from image coordinate frame using [this function](https://github.com/graspnet/graspnet-baseline/blob/main/utils/data_utils.py#L18).

> The depth can assume 4 values equal to hmax_list=[0.01,0.02,0.03,0.04], but I do not understand why this value should be "the distance from the origin of gripper coordinate frame to the gripper tip".

It is a definition, the translation of gripper tip is computed by the depth and grasp point.

> If I want to modify the code in order to take into account my gripper geometry, I should modify:
......
hmin and hmax_list.

hmax_list is related to the depths sampled in grasp labels, which should not be modified.

FrancescoRosa3, 2021-08-29:
Thank you, it seems clearer to me now. 
So, if I want to use this method for my experiments I have to modify only those variables that in the code refer to GRIPPER dimensions, is it correct?
These variables affect mainly the collision detection since the inference seems to be gripper independent, isn't it?
Thanks for your kindness.

chenxi-wang, 2021-08-30:
Yes, most of the gripper parameters are used in collision detection.

closed 2021-11-03 by chenxi-wang