graspnet/graspnet-baseline #25: Friction coefficients used during the evaluation

FrancescoRosa3, 2021-08-09:
Hi everyone.
I was running the evaluation procedure, and I would like to have some clarification:

1. In the published paper you said that the **AP** is computed as the average of the AP_u where u ranges from 0.2 to 1 with step of 0.2, but in the code, the friction coefficients range from 0.2 to 1.2, why is there this discrepancy?
2. Which is the set of coefficients used for computing the performance reported in the README?
3. Is the model obtained from the pre-trained weights the same as the one used for computing the reported performances?

Thanks for your kind.

chenxi-wang, 2021-08-11:
1. Sorry for the typo, it should be 1.2 in the paper. The detailed definition and evaluation protocol is defined in https://graspnet.net/evaluation.html.
2. The evaluation uses current graspnetAPI code.
3. Yes, the pretrained weights can reproduce the reported performance in README (not sure if there is a small difference).

FrancescoRosa3, 2021-08-11:
Thanks so much for the clarification.
Yes, there is a small difference but it is actually irrelevant.

closed 2021-08-11 by FrancescoRosa3