graspnet/graspnet-baseline #28: Problems with the generated rotation matrix

JingKangSun, 2021-08-20:
Hi @chenxi-wang,

I was trying to transplanted this algorithm to the real robotic arm.

I found that the rotation matrixs generated by the pretrained model have difference with the real camera frame.Specifically,I need to multiply a matrix to the left of the  rotation matrixs generated by the pretrained model,then I can get the correct matrixs under the camera frame.And the matrix T which I need to multiply left is:[0,0,1;1,0,0;0,1,0]

Blow is a instance of my question,the frame A is the camera frame,the frame B is the target frame generated by the pretrained model,the frame C is multiply T to the left of the frame B,and this is the frame corrspond to real camera frame:

[image]

Can you tell me why I need to multiply a matrix to the left of the  rotation matrixs generated by the pretrained model?Thanks!

Regards,
JingKang

chenxi-wang, 2021-08-30:
Could you provide specific examples for clearer explanations? I cannot figure out your meaning.

closed 2021-09-06 by JingKangSun

shineYuSong, 2022-12-08:
> Hi @chenxi-wang,
>
> I was trying to transplanted this algorithm to the real robotic arm.
>
> I found that the rotation matrixs generated by the pretrained model have difference with the real camera [...]

I am trying to apply this project to my real robotic arm too, have you finished? I was confused about the datainput that i only provide the rgb and depth information, the code in this project is locked with so much detial information, can you share your solution？