graspnet/graspnet-baseline #31: Realsesnse 435 Pointclouds quality issue

holst456, 2021-09-09:
Great article! I managed to get the code up and running and I am considering using it for my bachelor thesis in robotics.
My problem is that i cannot get nearly as good point-clouds as you have in your demo data. Did you use any filters or fused multi-view point clouds etc.?
E.g. this
[image]
Turns into this:
[image]

TuanHAnhVN, 2022-09-22:
> Great article! I managed to get the code up and running and I am considering using it for my bachelor thesis in robotics. My problem is that i cannot get nearly as good point-clouds as you have in [...]

Hello, did you solve it?

holst456, 2022-10-19:
> Hello, did you solve it?

Partially, yes. By improving light conditions and filtering outlier points.