graspnet/graspnet-baseline #35: 多显卡训练问题

LXZ1998-666, 2021-11-02:
您好，我在尝试使用多显卡训练的时候，设置参数如下：
`os.environ["CUDA_VISIBLE_DEVICES"] = '0,3'
net = nn.DataParallel(net.cuda())`
但是模型加载数据的时候出现错误：
`RuntimeError: Caught RuntimeError in replica 0 on device 0.`
请问作者在多显卡训练的时候是否遇到过这个问题？

chenxi-wang, 2021-11-03:
您好，我没有尝试过多卡训练，看起来可能是模型上一部分参数的device不对，可以检查一下

xiaozheng-liu, 2021-11-03:
感谢您的回答，我会在尝试修改代码；如果修改产生效果，我会向您回复的。

closed 2021-11-03 by xiaozheng-liu

xiaozheng-liu, 2021-11-03:
您好，在阅读论文的过程中，产生一些疑问：
关于评价精度AP，以 u = 0.8 为例 ，对应的抓取分数应该是 s =  1.1 -u = 0.3，假设排名前50的抓取位姿分数都是0.3，总的分数是15，为什么您论文中报告的是 **33.43/36.19**，我对此不太理解？AP是如何计算的？期待您的回复

xiaozheng-liu, 2021-11-03:
对于u=0.8时，夹爪预测的分数是  s = 0.3或者s = 0吗？在您的论文中，我看到Operation Network的损失函数是：
[image]
对于分数Sij的预测，我理解可以设计为分类损失函数，您为何选择回归预测呢？

Fang-Haoshu, 2021-11-03:
你好，抓取分数不是50*0.3这么算的，具体可以看我们论文中的阐述，或者网页的解释：https://graspnet.net/evaluation.html
分数的预测也可以使用分类损失，回归是为了兼容未来可能出现的其他度量。

xiaozheng-liu, 2021-11-03:
感谢您及时的回复，根据您的建议，我在论文中看到： **the percentage of true positive is more important.**，是否可以这样理解：在 u = 0.8 是，AP代表的是排名前50个抓取位姿中，真阳性的比例？比如说真阳性的比例是33.43%？

Fang-Haoshu, 2021-11-03:
您好，是这样的

xiaozheng-liu, 2021-11-03:
十分感谢您的回复，解答了疑惑，再次感谢您出色的工作！