graspnet/graspnet-baseline #37: Annotate grasp pose

qingweihk, 2021-11-29:
Thanks for your great job.Your annotation method is Brilliant.

I noticed your annotation method including two main parts:
1. 6D pose annotation
2. Grasp pose annotation

May I ask could you please provide detailed tools for annotating the second part: Grasp pose ?Cause we are helpless for this part annotation.
Thanks very much.

elevenjiang1, 2022-01-04:
yeah~
I know a tool named [LabelFusion](https://github.com/RobotLocomotion/LabelFusion) but the code is too old and difficult to use

Fang-Haoshu, 2022-11-11:
Hi, if you are interested in the object grasp pose annotation, you can fill out this form and I will invite you later on:  https://forms.gle/7o4f18UHsw55nQ256

ljc0815, 2024-03-31:
> 你好，如果你对物体抓取姿势标注感兴趣，你可以填写这个表格，我稍后会邀请你： https://forms.gle/7o4f18UHsw55nQ256

您好,我已经填写表格了,可以邀请下我吗?

zhengshuai1, 2024-12-18:
> Hi, if you are interested in the object grasp pose annotation, you can fill out this form and I will invite you later on: https://forms.gle/7o4f18UHsw55nQ256

Thanks for your great work! I have filled out the form, could you invite me?

JohnBrann, 2025-03-07:
I also filled out the form! Is this tool specifically the data in 'grasp_label' for each individual object? (points.npy, scores.npy, collisions.npy, and offsets.npy) all for their own respective object? Or is there another way to do this? All I really want to do is take a point cloud of a new object and generate potential grasps

Fang-Haoshu, 2025-03-08:
Invited

seok-jong, 2025-04-02:
Hi, I am interested in your project. I would also like to create my own dataset. Can you invite me? I have submitted the form above. Thank you.

Leon-0919, 2025-04-09:
Hey @Fang-Haoshu @chenxi-wang, thanks so much for the great work and your kind support to the community. 

I am also trying to train your awesome Graspnet-1 billion with **new data** in my domain like plastic containers. 

Thanks to your support, we have access to your annotation tool for a single object now. However, I have 2 issues at this stage. It would be invaluable if you could help check our problem. 

1. We are trying to run the code and generate sdf files. But we have an error saying **they are not valid sdf files**. Have you had any issues like this?
2. If we successfully add grasping poses to new objects and obviously share with the Graspnet community as you require. How do you generate valid poses for a scene?  I am reading your IJRR paper and it looks like we have another toolkit to align the objects in a scene from different viewpoints and then there is a collision detection and pose elimination process. 

Thanks for your patience with my questions and look forward to your reply!

All the best

[image]

Fang-Haoshu, 2025-04-21:
> Hi, I am interested in your project. I would also like to create my own dataset. Can you invite me? I have submitted the form above. Thank you.

invited

Edmolova, 2025-05-12:
> Hi, if you are interested in the object grasp pose annotation, you can fill out this form and I will invite you later on: https://forms.gle/7o4f18UHsw55nQ256

Thank you for your great work and warm heart. I would also like to create my dataset. Can you invite me? I have submitted the form above.
sincerely!

Fang-Haoshu, 2025-05-28:
invited @NoobTitan @Edmolova

FengMaverick, 2025-06-09:
> Hi, if you are interested in the object grasp pose annotation, you can fill out this form and I will invite you later on: https://forms.gle/7o4f18UHsw55nQ256

Hello, thank you for your contribution. I'm working on a task of grasping a non-cooperative target with a robotic arm. I'd like to learn more about the pose annotation for grasping and would like to use the annotation tool. I have submitted the form and hope to get your consent. Thank you!

Slonna, 2025-06-26:
> > Hi, I am interested in your project. I would also like to create my own dataset. Can you invite me? I have submitted the form above. Thank you.
> 
> invited

I have filled the form, can you invite me? Many thanks!

Edmolova, 2025-06-29:
> invited [@NoobTitan](https://github.com/NoobTitan) [@Edmolova](https://github.com/Edmolova)

Dear Fang blogger: I am very sorry, because I was on a business trip last week and failed to accept the invitation to graspnet/object-grasp-annotationrepository from you in time. I am really sorry to miss this opportunity. I wonder if it is convenient for you to send an invitation again? I really hope to continue to follow up on my studies and use relevant resources.

zhezixiansheng, 2025-07-30:
> 邀请 [@NoobTitan](https://github.com/NoobTitan) [@Edmolova](https://github.com/Edmolova)
您好 我已经填写了表格 希望得到您的邀请