graspnet/graspnet-baseline #38: How to measure the accuracy of estimation

watashihageteru, 2021-12-15:
I have a question about how to measure the estimation accuracy. In both training (train.py) and evaluation (test.py), the estimation accuracy seems to be calculated by loss.py. However, when I look at loss.py, I can't help but feel that it assumes that the answer is known before both training and evaluation. For example, the objectness_mask calculated by the compute_view_loss function is generated from the training data (objectness_label), but I think it should be generated from its own output (objectness_pred). Similarly, I thought the same could be said for the graspable_mask calculated by the compute_grasp_loss function. If I have misunderstood something, I would appreciate it if you could let me know.

Fang-Haoshu, 2022-01-06:
We compare the predicted results with the label to get the accuracy.

closed 2022-01-06 by Fang-Haoshu

1mingW, 2023-08-02:
In the unseen test dataset, where there is no ground truth for grasp poses, how did you compute the average precision?

chenxi-wang, 2023-08-04:
> In the unseen test dataset, where there is no ground truth for grasp poses, how did you compute the average precision?

@1mingW The ground truth is computed online during evaluation. You can refer to our paper (graspnet-1billion) for details.