graspnet/graspnet-baseline #39: Make grasp dataset in simulation

elevenjiang1, 2022-01-04:
Hello, it seems that all data you make is in real environment?
Will you make you data in simulation like pybullet or others like dex-net pipeline?
The dex-net pipeline means that all data is synthesis and robot can work in real environment, as far as i know, the competition OCRTOC, seems you have participated, may a good place to generate synthesis dataset, have you began to do so or?

Thank you~

Fang-Haoshu, 2022-01-06:
Hi, for using it in the simulation, you do not need to retrain the network. We directly used this repo and use the realsense model. The results is pretty good. Our model is robust in both realworld and simulation

Best

closed 2022-01-06 by Fang-Haoshu

elevenjiang1, 2022-01-06:
> Hi, for using it in the simulation, you do not need to retrain the network. We directly used this repo and use the realsense model. The results is pretty good. Our model is robust in both realworld and simulation
> 
> Best

Thank you for replay,however the reason why I want to retrain it in simulation is:
1. Objects are difficult to get (though can buy from network)
2. To compare same network with different objects or even different network with different objects

By the way, we test graspnet in OCRTOC competition, it is not robust in some scenes.
For example in scene 2-2-3,only two grasp for a single bow
[image]

Best

Fang-Haoshu, 2022-01-06:
I see. To generate data in simulation would be easy with 6D pose and any rendering tool. I would recommend pyrender.

The graspnet-baseline will produce dense prediction. For this case, I guess either the threshold is too high. or the coordination system is not set correctly.

In our updated version of the baseline, the performance is satisfactory and the demo is at: https://www.bilibili.com/video/BV17b4y1Z75f?zw It will also be available soon. Stay tuned!