graspnet/graspnet-baseline #43: Is this result right?

kaku2000, 2022-09-18:
My environment is :

OS：Ubuntu 22.04.1 LTS
CUDA：11.7     sm_86
Pytorch：1.8.0

Since I couldn't successfully install pytorch_knn_cuda repo, I installed https://github.com/unlimblue/KNN_CUDA this one instead. And simply replaced "knn" function by the "KNN". 

As a result, when running demo.py with the pretrained checkpoint, no matter using "rs" or "kn",  I get this prediction:

[image]

Is this right? Or should I figure out how to install the old pytorch_knn_cuda ?

Thanks.

chenxi-wang, 2022-09-20:
Hi, the result is incorrect. The knn operator should be installed using the code in ``knn`` folder (in this repo) instead of pytorch_knn_cuda repo.

kaku2000, 2022-09-20:
> Hi, the result is incorrect. The knn operator should be installed using the code in `knn` folder (in this repo) instead of pytorch_knn_cuda repo.

Oh, it is right, after turning the pointcloud 180 degree LOLLL

closed 2022-09-20 by kaku2000