graspnet/graspnet-baseline #44: AttributeError: 'ColorVisuals' object has no attribute 'crc'

zhaozj89, 2022-10-11:
I tried to use `sh command_test.sh` to test a trained model, but it reported an error in loading dexnet model, as shown in the following. I would like to inquire if you have any ideas to solve this problem. Thanks very much.

```
Loading data path and collision labels...: 100%|████████████████████████| 90/90 [00:00<00:00, 219.96it/s]
23040
23040
-> loaded checkpoint logs/log_rs/checkpoint.tar (epoch: 12)
Loading data path...: 100%|█████████████████████████████████████████████| 90/90 [00:00<00:00, 205.90it/s]
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/amax/miniconda3/envs/py37/lib/python3.7/multiprocessing/pool.py", line 121, in worker
    result = (True, func(*args, **kwds))
  File "/home/amax/zzhaoao/Grasp/graspnetAPI/graspnetAPI/graspnet_eval.py", line 121, in eval_scene
    model_list, dexmodel_list, _ = self.get_scene_models(scene_id, ann_id=0)
  File "/home/amax/zzhaoao/Grasp/graspnetAPI/graspnetAPI/graspnet_eval.py", line 52, in get_scene_models
    dexmodel = pickle.load(f)
AttributeError: 'ColorVisuals' object has no attribute 'crc'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "test.py", line 119, in <module>
    evaluate()
  File "test.py", line 113, in evaluate
    res, ap = ge.eval_all(cfgs.dump_dir, proc=cfgs.num_workers)
  File "/home/amax/zzhaoao/Grasp/graspnetAPI/graspnetAPI/graspnet_eval.py", line 305, in eval_all
    res = np.array(self.parallel_eval_scenes(scene_ids = list(range(100, 190)), dump_folder = dump_folder, proc = proc))
  File "/home/amax/zzhaoao/Grasp/graspnetAPI/graspnetAPI/graspnet_eval.py", line 231, in parallel_eval_scenes
    scene_acc_list.append(res.get())
  File "/home/amax/miniconda3/envs/py37/lib/python3.7/multiprocessing/pool.py", line 657, in get
    raise self._value
AttributeError: 'ColorVisuals' object has no attribute 'crc'
```

closed 2022-10-13 by zhaozj89

shineYuSong, 2022-11-21:
so how to solve it, i occur to this either.

Fang-Haoshu, 2022-11-24:
I reproduce the error at my side and found that it is the trimesh version problem. An older version of trimesh<=3.15.1 fixes this problem.

Fang-Haoshu, 2022-11-25:
The official dex models are also updated. Thus you can either use the older version of trimesh, or redownload the dex model from the website.