graspnet/graspnet-baseline #47: AttributeError: 'ColorVisuals' object has no attribute 'crc'

shineYuSong, 2022-11-21:


Fang-Haoshu, 2022-11-22:
Hi, we haven't met that issue before. Perhaps you can try to search for that on Google. I found some related issues like https://github.com/mikedh/trimesh/issues/786 but not sure if the solution works.

Fang-Haoshu, 2022-11-22:
perhaps try some other trimesh version

xiaoxingrunning, 2022-11-23:
I have encountered the same problem. Have you solved it？

shineYuSong, 2022-11-24:
> perhaps try some other trimesh version

I dont use trimesh package, and here is all my packages,but i have 
open3d  0.16.0, and the  error trace was:

Traceback (most recent call last):
  File "**/anaconda3/envs/graspp/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "**/anaconda3/envs/graspp/lib/python3.8/site-packages/graspnetAPI/graspnet_eval.py", line 119, in eval_scene
    model_list, dexmodel_list, _ = self.get_scene_models(scene_id, ann_id=0)
  File "**/anaconda3/envs/graspp/lib/python3.8/site-packages/graspnetAPI/graspnet_eval.py", line 50, in get_scene_models
    dexmodel = pickle.load(f)
AttributeError: 'ColorVisuals' object has no attribute 'crc'

Fang-Haoshu, 2022-11-24:
perhaps try an older pickle version?
https://github.com/graspnet/graspnetAPI/blob/master/graspnetAPI/graspnet_eval.py#L48-L50
A quick workaround is to comment the if condition here. The evaluation code will try to generate the dexmodel for evaluation.

Fang-Haoshu, 2022-11-24:
I reproduce the error at my side and found that it is indeed the trimesh version problem. An older version of trimesh<=3.15.1 fix this problem.

closed 2022-11-24 by Fang-Haoshu

shineYuSong, 2022-11-25:
> I reproduce the error at my side and found that it is indeed the trimesh version problem. An older version of trimesh<=3.15.1 fix this problem.

yes i have fixed it thanks a lot~

Fang-Haoshu, 2022-11-25:
The official dex models are also updated. Thus you can either use the older version of trimesh, or redownload the dex model from the website.

ibtehajali67, 2022-11-26:
> I reproduce the error at my side and found that it is indeed the trimesh version problem. An older version of trimesh<=3.15.1 fix this problem.

thank you it solved my problem