graspnet/graspnet-baseline #50: Question about the in-plane rotation

hooyahliu, 2023-02-16:
Thank you for your excellent research achievements.
I have a question about the in-plane rotation.
Where is the starting position of the angle defined？
If there is an explanation in paper, please remind me, thank you!

感谢您出色的研究成果。
有关于in-plane rotation我有一个问题。
请问in-plane rotation的初始位置，也就是0度位置定义在何处？
如果在文章中有说明，麻烦您提示一下，谢谢！

chenxi-wang, 2023-02-23:
The definition is [here](https://github.com/graspnet/graspnet-baseline/blob/main/utils/loss_utils.py#L68).

closed 2023-02-23 by chenxi-wang

dingthuang, 2024-05-02:
> The definition is [here](https://github.com/graspnet/graspnet-baseline/blob/main/utils/loss_utils.py#L68).

Hello, may I ask a question. When loading the dataset, labels for individual objects will be rotated. Only the view has been modified, but the angle has not been modified. This seems to cause issues with the label angle.

Simply put, rotation 1 is represented by view1 and angle1, while rotation 2 is represented by view2 and angle2. Rotate 1 through the rotation matrix to rotate 1 '(Rotate 2). View 1 rotates the matrix to view 1 (view 2). But rotation 1 'cannot be represented by view1' and angle1.