graspnet/graspnet-baseline #51: pointnet undefined symbol error

NickNair-ari, 2023-02-26:
Trying to run  the demo.py on cuda 11.8, torch 1.10.0 on Ubuntu 22.04 

This is the terminal output 
## Command :
```
python3 demo.py --checkpoint_path checkpoint-kn.tar 
```
## Output 
```
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
Traceback (most recent call last):
  File "/home/nicknair/graspnet-baseline/pointnet2/pointnet2_utils.py", line 26, in <module>
    import pointnet2._ext as _ext
ImportError: /home/nicknair/anaconda3/envs/py3-mink/lib/python3.8/site-packages/pointnet2-0.0.0-py3.8-linux-x86_64.egg/pointnet2/_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor6deviceEv

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "demo.py", line 22, in <module>
    from graspnet import GraspNet, pred_decode
  File "/home/nicknair/graspnet-baseline/models/graspnet.py", line 18, in <module>
    from backbone import Pointnet2Backbone
  File "/home/nicknair/graspnet-baseline/models/backbone.py", line 14, in <module>
    from pointnet2_modules import PointnetSAModuleVotes, PointnetFPModule
  File "/home/nicknair/graspnet-baseline/pointnet2/pointnet2_modules.py", line 21, in <module>
    import pointnet2_utils
  File "/home/nicknair/graspnet-baseline/pointnet2/pointnet2_utils.py", line 29, in <module>
    raise ImportError(
ImportError: Could not import _ext module.
Please see the setup instructions in the README: https://github.com/erikwijmans/Pointnet2_PyTorch/blob/master/README.rst

```

Fang-Haoshu, 2023-03-05:
Hi, did you successfully install the pointnet2 operators? Are there any errors during installation?

NickNair-ari, 2023-03-05:
I was able to install pointnet2. The problem was with having the correct pytorch version compatible with my cuda version. Later on, I had trouble installing knn operators. I did a `pip install .` instead of `python3 setup.py install`, and it somehow worked.
Right now, everything is up and running.
I have two other issues though 
1. When trying to run the model through my own data, I find that much of the grasp poses are generated for the edges of the pointcloud, like in the image below. I'd appreciate it if you could suggest anything in the utils or any post-processing that may help with this issue.
[image]
2. I wanted to try out this repo also : [anygrasp_sdk](https://github.com/graspnet/anygrasp_sdk), but that seems to require a license. I filled out a google-form pertaining to it but got no response. It'd be a great help if that could be arranged.

All in all, this a great project, and I found it really interesting to run and experiment with.

chenxi-wang, 2023-03-06:
Hi, you can set a mask to input points like the one in demo.py, or set a workspace to filter the output grasps.
Do you receive the license now? If not, provide me your email so that I can have a check.

NickNair-ari, 2023-03-15:
I haven't received any license yet. You may mail it to : nikhilnicky972@gmail.com

Also, the mask only helps to a certain degree. I wonder if there is a way to filter the grasp points after passing it through the model. Even with a mask the pointcloud will still have edges the model generates grasp poses for.

chenxi-wang, 2023-03-17:
> I haven't received any license yet. You may mail it to : [nikhilnicky972@gmail.com](mailto:nikhilnicky972@gmail.com)
> 
> Also, the mask only helps to a certain degree. I wonder if there is a way to filter the grasp points after passing it through the model. Even with a mask the pointcloud will still have edges the model generates grasp poses for.

Sorry, I did not find your email in received application. Maybe you need to apply again.

You can apply a mask directly to your output grasps (use translation/rotation limitation) or ''expand'' your table plane of the input cloud so that the edges are far from objects.

closed 2023-04-25 by chenxi-wang

elenacliu, 2023-08-10:
@sdjdlhg I have the same question

Fang-Haoshu, 2023-08-16:
> I was able to install pointnet2. The problem was with having the correct pytorch version compatible with my cuda version. Later on, I had trouble installing knn operators. I did a `pip install .` [...]

@elenacliu @sdjdlhg The author have already mentioned his solution.