graspnet/graspnet-baseline #57: How come the output of this model is stochastic?

cx3307, 2023-05-09:
How come the output of this model is stochastic? When the same image is input twice, the results obtained are different.
This is my input rgb image:
[image]
This is the first output:
[image]
This is the second output:
[image]

chenxi-wang, 2023-05-11:
Did you sample the points randomly when creating the point cloud?

Fang-Haoshu, 2023-07-22:
In the above images, the RGBD images are not the same for the inputs.

closed 2023-07-22 by Fang-Haoshu