graspnet/graspnet-baseline #58: How did you avoid the collison?

ZIXINLinsztu, 2023-05-18:
After I input the RGB and Depth image into the demo.py, I got the prediction data. Then I transform the coordinate so that my robot can move to the target pose, but it would collide the object,. So I wonder how did you avoid the collision between gripper and object in your experiment?

Fang-Haoshu, 2023-05-18:
Hi, we first move the gripper to a pose that is 10cm backward along the approaching direction of the predicted grasp pose, and then make the gripper moves to the predicted grasp pose along the approaching direction.

ZIXINLinsztu, 2023-05-19:
> Hi, we first move the gripper to a pose that is 10cm backward along the approaching direction of the predicted grasp pose, and then make the gripper moves to the predicted grasp pose along the approaching direction.

It's helpful for me, thanks!

closed 2023-05-19 by ZIXINLinsztu

WentangChen, 2023-07-04:
> After I input the RGB and Depth image into the demo.py, I got the prediction data. Then I transform the coordinate so that my robot can move to the target pose, but it would collide the object,. So I wonder how did you avoid the collision between gripper and object in your experiment?

hi there, I also input the RGB and Depth image into the demo.py, but i cannot get the prediction data. I wonder if you could offer your code for me to learn about it? Thanks.