graspnet/graspnet-baseline #64: 如何将算法部署在搭配realsense的机械臂上？

WentangChen, 2023-07-04:
您好！
首先十分感谢您优秀的工作！
我打算将你的算法应用在搭配了realsense的机械臂上。有几个问题想问一下：
1. 我在用realsense保存的rgb和depth图像替换demo.py中使用的doc/example_data中的图像，但是得到很奇怪的结果：
[image]
我排查应该是Depth图像的问题，请问depth.png图像的获取有什么要求吗？或者有什么其他的要求？
2. workspace_mask.png如何生成？可以直接沿用提供的图片吗？

WentangChen, 2023-07-05:
我更改了depth图片的数据格式后，得到以下图像：
[image]
我这里深度没有显示出来，请问可能会是什么原因呀？

WentangChen, 2023-07-06:
> depth直接保存读取出的数据就可以，可以检查一下是不是在读取或者保存的时候出现了问题 workspace mask是根据实验需要自己划定的，demo里只是个示例 更改depth图片的数据格式是什么意思？

1. 请问graspnet数据集在每个scenes保存深度图像的工具有提供吗？我直接读取出来的depth数据和数据集给的depth.png不一样
2. workspace.mask的问题我已经解决了，感谢！
3. 就是在保存之前： depth data = np.asanyarray(aligned_depth_frame.get _data()， dtype=np.float32) 选择dtype=np.float32

chenxi-wang, 2023-07-06:
> 1. 请问graspnet数据集在每个scenes保存深度图像的工具有提供吗？我直接读取出来的depth数据和数据集给的depth.png不一样
> 2. workspace.mask的问题我已经解决了，感谢！
> 3. 就是在保存之前： depth data = np.asanyarray(aligned_depth_frame.get _data()， dtype=np.float32) 选择dtype=np.float32

depth图像最好保存成 uint16 格式，你保存以后可以直接读出来看看数组的深度值对不对

WentangChen, 2023-07-06:
> > 1. 请问graspnet数据集在每个scenes保存深度图像的工具有提供吗？我直接读取出来的depth数据和数据集给的depth.png不一样
> > 2. workspace.mask的问题我已经解决了，感谢！
> > 3. 就是在保存之前： depth data = np.asanyarray(aligned_depth_frame.get _data()， dtype=np.float32) 选择dtype=np.float32
> 
> depth图像最好保存成 uint16 格式，你保存以后可以直接读出来看看数组的深度值对不对

非常感谢你这么快的恢复！
我保存为uint16格式后运行得到下面的结果：
[image]
出现这个现象的原因可能是什么？
请问除了color.png和depth.png、相机内参需要提供并更改之外，我还需要提供/更改什么地方吗？

chenxi-wang, 2023-07-07:
你是想问为什么出现这么多噪声点吗，这个跟你光照环境，相机位置都有关系，你可以用realsense viewer调一下成像效果

WentangChen, 2023-07-07:
> 你是想问为什么出现这么多噪声点吗，这个跟你光照环境，相机位置都有关系，你可以用realsense viewer调一下成像效果

好的！我拿到设备再试一下。请问获取depth数据前是需要一个处理的过程吗？
另外是不是我预测了之后就可以应用得到的抓取姿态尝试抓取了？

chenxi-wang, 2023-07-07:
> 请问获取depth数据前是需要一个处理的过程吗？

不需要，跟rgb坐标align就行

> 另外是不是我预测了之后就可以应用得到的抓取姿态尝试抓取了？

是的

WentangChen, 2023-07-07:
> > 请问获取depth数据前是需要一个处理的过程吗？
> 
> 不需要，跟rgb坐标align就行
> 
> > 另外是不是我预测了之后就可以应用得到的抓取姿态尝试抓取了？
> 
> 是的

十分感谢！我再试一下看看效果如何

closed 2023-07-08 by Fang-Haoshu

chenxi-wang, 2023-07-19:
我没搞懂你的意思，你是想说可视化吗

WentangChen, 2023-07-19:
> 我没搞懂你的意思，你是想说可视化吗

是的，实现实时的可视化