graspnet/graspnet-baseline #69: This result is fantastic!

YKJK, 2023-07-23:
1.Can you give me some advice on how to modify this model for real-time display of grasp poses?

2.Will my 3D camera, which produces black and white point clouds, affect the results I see?

3.I am currently unable to successfully generate the file format (like checkpoint-kn.tar).

If you have seen this, please reply as soon as possible.

chenxi-wang, 2023-07-24:
1. You can refer to open3d official documents and their visualization demo.
2. No. This model only needs point clouds as input.
3. This can be done with pytorch function.
4. Please provide your email filled in the form and I will have a check.

YKJK, 2023-07-24:
Thank you for your response. I will try again!!
I will ask you if I have any questions.
I have just re-submitted the form. Please take a look again.

chenxi-wang, 2023-07-25:
Your license has been sent a week ago. Did you check your mail box?

closed 2023-08-04 by chenxi-wang