graspnet/graspnet-baseline #72: How to run this project？

ChengXIZhang666, 2023-09-01:
The following are the errors reported when installing knn：
vision.obj : error LNK2001: 无法解析的外部符号 "public: long * __cdecl at::Tensor::data_ptr<long>(void)const " (??$data_ptr@J@Tensor@at@@QEBAPEAJXZ)
vision.obj : error LNK2001: 无法解析的外部符号 "struct THCState * state" (?state@@3PEAUTHCState@@EA)
build\lib.win-amd64-3.6\knn_pytorch\knn_pytorch.cp36-win_amd64.pyd : fatal error LNK1120: 2 个无法解析的外部命令
error: command 'F:\\VSsoft\\Visual Studio IDE\\VC\\Tools\\MSVC\\14.36.32532\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

chenxi-wang, 2023-09-04:
That is probably caused by incorrect cuda version or pytorch version. Besides, we did not test the code on Windows. Linux is recommended.

ChengXIZhang666, 2023-09-04:
alright  and thanks

closed 2023-09-08 by chenxi-wang