graspnet/graspnet-baseline #73: 关于knn编译报错

fuzhao123232, 2023-09-04:
[图片]请问是必须pytorch1.7吗，高版本的pytorch不能支持吗？

fuzhao123232, 2023-09-04:
[图片]

fuzhao123232, 2023-09-04:
[image]

chenxi-wang, 2023-09-04:
如果你用了较高版本的pytorch，需要修改一些接口才能编译，类似 https://stackoverflow.com/questions/72988735/replacing-thc-thc-h-module-to-aten-aten-h-module

closed 2023-09-04 by fuzhao123232

nikepupu, 2023-10-07:
Can you share the modified file @fuzhao123232 ?

quanfeifan, 2024-03-13:
请问你解决这个问题了吗

YuyangLee, 2024-05-03:
Hi, the following procedures work for me. However, I'm not an expert in CUDA and C++ so it would be great if someone professional could refine the solution.

### Modify `knn/src/cuda/vision.h`

First, go to `knn/src/cuda/vision.h` and comment this line:

```cpp
#include <THC/THC.h>
```

Append the following codes after this line:

```cpp
#include <ATen/cuda/CUDAContext.h>
#include <ATen/cuda/CUDAEvent.h>
```

Also in this file we need to change several APIs:

```cpp
// float *dist_dev = (float*)THCudaMalloc(state, ref_nb * query_nb * sizeof(float));
// Change this to:
float *dist_dev = (float*)c10::cuda::CUDACachingAllocator::raw_alloc(ref_nb * query_nb * sizeof(float));
```

```cpp
// THCudaFree(state, dist_dev);
// Change this to:
c10::cuda::CUDACachingAllocator::raw_delete(dist_dev);
```

**About the last change, I am not sure what's the substitute for `THError` so I temporarily use `return 0` here. I believe there should be a proper solution for this.**

```cpp
cudaError_t err = cudaGetLastError();
if (err != cudaSuccess)
{
    printf("error in knn: %s\n", cudaGetErrorString(err));
    return 0;
    // THError("aborting");
}
```

### Modify `knn/src/knn.h`

Go to `knn/src/knn.h` and comment these lines:

```cpp
#include <THC/THC.h>
extern THCState *state;
```

### Install the Package

Finally we can install the package `knn/`, and the `demo.py` works fine for me.

### Reference

- https://github.com/pytorch/pytorch/issues/72807#issuecomment-1699371013

airobot1024, 2024-06-21:
> Hi, the following procedures work for me. However, I'm not an expert in CUDA and C++ so it would be great if someone professional could refine the solution.
> 
> ### Modify `knn/src/cuda/vision.h`
> First, go to `knn/src/cuda/vision.h` and comment this line:
> 
> ```c++
> #include <THC/THC.h>
> ```
> 
> Append the following codes after this line:
> 
> ```c++
> #include <ATen/cuda/CUDAContext.h>
> #include <ATen/cuda/CUDAEvent.h>
> ```
> 
> Also in this file we need to change several APIs:
> 
> ```c++
> // float *dist_dev = (float*)THCudaMalloc(state, ref_nb * query_nb * sizeof(float));
> // Change this to:
> float *dist_dev = (float*)c10::cuda::CUDACachingAllocator::raw_alloc(ref_nb * query_nb * sizeof(float));
> ```
> 
> ```c++
> // THCudaFree(state, dist_dev);
> // Change this to:
> c10::cuda::CUDACachingAllocator::raw_delete(dist_dev);
> ```
> 
> **About the last change, I am not sure what's the substitute for `THError` so I temporarily use `return 0` here. I believe there should be a proper solution for this.**
> 
> ```c++
> cudaError_t err = cudaGetLastError();
> if (err != cudaSuccess)
> {
>     printf("error in knn: %s\n", cudaGetErrorString(err));
>     return 0;
>     // THError("aborting");
> }
> ```
> 
> ### Modify `knn/src/knn.h`
> Go to `knn/src/knn.h` and comment these lines:
> 
> ```c++
> #include <THC/THC.h>
> extern THCState *state;
> ```
> 
> ### Install the Package
> Finally we can install the package `knn/`, and the `demo.py` works fine for me.
> 
> ### Reference
> * [Missing headers in ATen/cuda/DeviceUtils.cuh pytorch/pytorch#72807 (comment)](https://github.com/pytorch/pytorch/issues/72807#issuecomment-1699371013)

Great! Many thanks. According your answer, I resolve this issue.
Below "Also in this file we need to change several APIs:",  the remained modification is belonged to  **knn/src/knn.h.**

liangyijian1, 2025-04-15:
> 您好，以下过程适用于我。但是，我不是 CUDA 和 C++ 方面的专家，所以如果有人可以改进解决方案，那就太好了。
> 
> ### Modify `knn/src/cuda/vision.h`
> 首先，转到 `knn/src/cuda/vision.h` 并注释这行：
> 
> #include <THC/THC.h>
> 在此行后附加以下代码：
> 
> #include <ATen/cuda/CUDAContext.h>
> #include <ATen/cuda/CUDAEvent.h>
> 此外，在此文件中，我们需要更改几个 API：
> 
> // float *dist_dev = (float*)THCudaMalloc(state, ref_nb * query_nb * sizeof(float));
> // Change this to:
> float *dist_dev = (float*)c10::cuda::CUDACachingAllocator::raw_alloc(ref_nb * query_nb * sizeof(float));
> // THCudaFree(state, dist_dev);
> // Change this to:
> c10::cuda::CUDACachingAllocator::raw_delete(dist_dev);
> **关于最后一次更改，我不确定 `THError` 的替代品是什么，所以我在这里暂时使用 `return 0`。我相信应该有一个合适的解决方案。**
> 
> cudaError_t err = cudaGetLastError();
> if (err != cudaSuccess)
> {
>     printf("error in knn: %s\n", cudaGetErrorString(err));
>     return 0;
>     // THError("aborting");
> }
> ### Modify `knn/src/knn.h`
> 转到 `knn/src/knn.h` 并注释这些行：
> 
> #include <THC/THC.h>
> extern THCState *state;
> ### Install the Package
> 最后，我们可以安装 `knn/` 包，`demo.py` 对我来说工作正常。
> 
> ### Reference
> * [Missing headers in ATen/cuda/DeviceUtils.cuh pytorch/pytorch#72807 (comment)](https://github.com/pytorch/pytorch/issues/72807#issuecomment-1699371013)

I encountered an error "error: 'c10::cuda::CUDACachingAllocator' has not been declared" when installing the latest version, how should I solve it?

```
(sim1) yijian@yiliang:~/repo/graspnet-baseline/knn$ python setup.py install
running install
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
[... 111 lines cut ...]
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/Context.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader_options.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader/base.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader/stateful.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/all.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/extension.h:4,
                 from /home/yijian/repo/graspnet-baseline/knn/src/cpu/vision.h:2,
                 from /home/yijian/repo/graspnet-baseline/knn/src/knn.h:2,
                 from /home/yijian/repo/graspnet-baseline/knn/src/vision.cpp:1:
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/core/TensorBody.h:268:30: note: declared here
   DeprecatedTypeProperties & type() const {
                              ^~~~
In file included from /home/yijian/repo/graspnet-baseline/knn/src/vision.cpp:1:0:
/home/yijian/repo/graspnet-baseline/knn/src/knn.h:33:42: error: ‘c10::cuda::CUDACachingAllocator’ has not been declared
     float *dist_dev = (float*)c10::cuda::CUDACachingAllocator::raw_alloc(ref_nb * query_nb * sizeof(float));
                                          ^~~~~~~~~~~~~~~~~~~~
/home/yijian/repo/graspnet-baseline/knn/src/knn.h:39:16: error: ‘c10::cuda::CUDACachingAllocator’ has not been declared
     c10::cuda::CUDACachingAllocator::raw_delete(dist_dev);
                ^~~~~~~~~~~~~~~~~~~~
error: command '/usr/bin/g++' failed with exit code 1
```