graspnet/graspnet-baseline #74: Question about grasping single thin object

baoyufuyou, 2023-09-04:
Hi guys,
I found that when planning grasping for single thin object, there is always no candidates. Could I ask, it that normal? Thanks in advance.

chenxi-wang, 2023-09-05:
How thin is the object? Did you visualize and check the input point cloud?

baoyufuyou, 2023-09-05:
[image]
The thin object is the HDMI cable with height about 1cm. I already checked the point cloud. When there are lots of cables in the scene, grasp planning could be done successfully. But when the cables are placed flat on the table, there are totally no candidates.

chenxi-wang, 2023-09-05:
You mean multiple cables are placed flat but there are no candidates?

baoyufuyou, 2023-09-05:
@chenxi-wang  Yes. For the scene with multiple cables or single cable placed flat on table, there is both no candidate.

chenxi-wang, 2023-09-06:
That's weird. The most likely explanation is that cables are treated as part of the table. But I'm not sure. Perhaps you can try anygrasp sdk.
Btw, the machine feature id you provided seems incorrect. I've sent you an email several days ago.

baoyufuyou, 2023-09-18:
@chenxi-wang Thanks a lot! I will try to submit the machine feature and try anygrasp sdk. Close this ticket and I will update the status later afterwards.

closed 2023-09-18 by baoyufuyou