graspnet/graspnet-baseline #75: Questions about the scene 6DOF pose annotation.

qiaoyo, 2023-09-05:
Hi, 
  Thanks for your great work and detailed API tool. I have received the Grasp pose annotation tool invitation. I noticed that it's basically related to PointnetGPD and integrated into the GraspNetAPI. 
  Besides the object pose annotation, the 6DOF pose annotation is also an important part in your great work. In your paper, this part is described in this way "we only need to annotate 6D poses for the first frame of each scene." I want to know how this part is achieved. Some tools I have known is the Elasticfusion and Labelfusion. The former is to rebuild the scene pointcloud, the latter is to annotate the 6DOF pose. Could u share the 6DOF annotation way of the first frame? Sincerely, thanks very much.

chenxi-wang, 2023-09-14:
Hi, we made a tool to render the objects to the image, in which way we can visualize and adjust 6dof poses easily.

closed 2023-09-14 by chenxi-wang

willieChen58, 2023-11-10:
> Hi, we made a tool to render the objects to the image, in which way we can visualize and adjust 6dof poses easily.

@chenxi-wang Hi, could you provide with the tools you made to annotate 6D poses? Because our scientific research project needs to use 6D pose estimation and grasp, we need to label the dataset of pose estimation, and there is no mature method of annotating the dataset, so I would like to ask if you can provide this annotation tool?

Leon-0919, 2025-04-11:
@chenxi-wang Hi Chenxi, do we have any further updates on this issue? Is this tool confidential or can be open-sourced? Thanks for your reply!

Edmolova, 2025-05-12:
> [@chenxi-wang](https://github.com/chenxi-wang) Hi, could you provide with the tools you made to annotate 6D poses? Because our scientific research project needs to use 6D pose estimation and grasp, [...]

Hello, we also have similar troubles. After reading the paper, I haven't found the answer for two days. I couldn't find the annotation tool for the first frame. I don't know if it can be made public. If you have the answer, could you please tell me? I sincerely seek this tool.

Edmolova, 2025-05-12:
> D
We noticed that you received the grasping pose annotation tool. Could you please share it with us? We are also facing a similar issue. After reading the paper, I have been looking for it for two days， but still haven't found the annotation tool for the first frame. I don't know if it can be made public. If you know the answer, please let me know. I sincerely hope to get this tool.

> n