graspnet/graspnet-baseline #79: obj ids in different split set

xins981, 2023-10-01:
Hi @chenxi-wang , can you provide obj_id lists for train, similar, unseen scene or any hint to seek it? I can't find information in official web-site and this repo. thanks advance for your any help!

chenxi-wang, 2023-10-09:
You can find a object_id_list.txt in each scene folder, which contains the objects that appear (0-indexed)

Fang-Haoshu, 2023-10-20:
@xins981 You can also use this function in the graspnetAPI to obtain obj ids for different scenes: https://github.com/graspnet/graspnetAPI/blob/master/graspnetAPI/graspnet.py#L219

closed 2023-10-20 by Fang-Haoshu

xins981, 2023-10-20:
Thanks for your clues! In fact, I want to know all object ids of the scene 0-100, 100-130, 130-160, 160-190 if you have statistics, respectively.