graspnet/graspnet-baseline #84: run demo error

MoonBunnyZZZ, 2023-12-12:
During run common_demo.sh, there is ImportError:

ImportError: numpy.core.multiarray failed to import (auto-generated because you didn't call 'numpy.import_array()' after cimporting numpy; use '<void>numpy._import_array' to disable if you are certain you don't need it).

It seems like about grasp_nms package. How to fix it?

MoonBunnyZZZ, 2023-12-12:
> During run common_demo.sh, there is ImportError:
> 
> ImportError: numpy.core.multiarray failed to import (auto-generated because you didn't call 'numpy.import_array()' after cimporting numpy; use 'numpy._import_array' to disable if you are certain you don't need it).
> 
> It seems like about grasp_nms package. How to fix it?

I  put 'numpy.import_array()' after cimporting numpy in grasp_nms.pyx.The error don't occur.But I don't kown whether this is the only right way.

chenxi-wang, 2023-12-13:
Maybe it's due to numpy version?

MoonBunnyZZZ, 2023-12-13:
> Maybe it's due to numpy version?

which version in your python env?In my env, it's 1.19.0.

chenxi-wang, 2023-12-13:
I didn't remember the version. You can try upgrading numpy directly.

MoonBunnyZZZ, 2023-12-14:
> I didn't remember the version. You can try upgrading numpy directly.

I install the newest version numpy first.But there is error about "no.float" which is deprecated in version 1.20 .So I try version 1.19.

MoonBunnyZZZ, 2023-12-14:
By the way, what is the gray point in point cloud generated by running demo .py?What does it mean?

chenxi-wang, 2023-12-15:
> By the way, what is the gray point in point cloud generated by running demo .py?What does it mean?

Could you provide more details about "gray point"?

MoonBunnyZZZ, 2023-12-20:
> > By the way, what is the gray point in point cloud generated by running demo .py?What does it mean?
> 
> Could you provide more details about "gray point"?

sorry for the delay
the gray point is in the top-right corner
[image]

chenxi-wang, 2024-01-09:
Could it be the origin of the camera coordinates?

captain316, 2024-01-09:
> I put 'numpy.import_array()' after cimporting numpy in grasp_nms.pyx.The error don't occur.But I don't kown whether this is the only right way.

but,  where is grasp_nms.pyx? I can't find it.

15138676358, 2024-01-12:
The function "nms" is aimed at non-maximum suppression. I commented out the line 108 [e.g. gg.nms()] of the demo.py file directly, which seems to have no great effect on the results of the example.

closed 2024-05-07 by chenxi-wang

briller-zzj, 2024-05-08:
> I put 'numpy.import_array()' after cimporting numpy in grasp_nms.pyx.The error don't occur.But I don't kown whether this is the only right way.

I'm also getting this error, but I can't find where the “grasp_nms.sh” file is, how did you fix it, could you tell me please?