graspnet/graspnet-baseline #88: what means of "assign views"?

xins981, 2024-01-11:
https://github.com/graspnet/graspnet-baseline/blob/fb2f089a9515d2a5e433b28b53827c1c914437f2/utils/label_generation.py#L60
In my opinion, grasp_views_trans_ is the viewpoints in camera frame, rather grasp_views_ ones in object or world frame. What is the goal of knn operation?

chenxi-wang, 2024-01-17:
The viewpoints generated from the object coordinates cannot be strictly matched to the ones from the camera coordinates one by one.  So the grasp labels at one viewpoint generated from the camera coordinates (in training) are estimated by searching the nearest viewpoints from the object coordinates, where the original labels are in.

feixyz10, 2024-11-23:
@xins981 Hello, I'm also very confused about the use of KNN here. The author's response above did not answer my confusion. I want to ask if you have finally figured it out why KNN is used? I would be very grateful if you can tell.