graspnet/graspnet-baseline #91: 自定义workspace_mask

du-xinyi, 2024-01-15:
如何自定义workspace_mask，我使用自定义的二值图取代workspace_mask后，会出现
```
    cloud_sampled = cloud_masked[idxs]
numpy.core._exceptions.MemoryError: Unable to allocate 732. GiB for an array with shape (20000, 1280, 1280, 3) and data type float64
```

du-xinyi, 2024-01-15:
[image]

du-xinyi, 2024-01-15:
> ![mask](https://private-user-images.githubusercontent.com/73220784/296750276-bb8e9aed-27ba-4ef6-99ee-3b7b2dca4d7f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

这是我设置的掩膜

xinchaosong, 2024-01-17:
Is your `workspace_mask` a boolean matrix? It must be a boolean matrix, not a numeric matrix.

chenxi-wang, 2024-01-17:
在mask过后，点云大小应该是N x 3，根据提示的信息，你在masking这步出了问题，可以检查一下mask是不是布尔值

closed 2024-01-17 by chenxi-wang