graspnet/graspnet-baseline #94: run command_demo.sh error

TheFive666, 2024-02-02:
I encountered an error while running the demo program. Could you please help me take a look? The GPU is GNVDIA3060, and my running environment is CUDA11, cudnnv8.7.0, pytorch 2.2.0+cu118, and python 3.8.8，I can use Open3D to display the point cloud normally
[image]

The error is as follows:

(base) w@w-Dell-G16-7620:~/graspnet-baseline-main$ sh command_demo.sh 
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
-> loaded checkpoint logs/log_kn/checkpoint-rs.tar (epoch: 18)
Traceback (most recent call last):
  File "demo.py", line 125, in <module>
    demo(data_dir)
  File "demo.py", line 118, in demo
    gg = get_grasps(net, end_points)
  File "demo.py", line 96, in get_grasps
    end_points = net(end_points)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/w/graspnet-baseline-main/models/graspnet.py", line 71, in forward
    end_points = self.view_estimator(end_points)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/w/graspnet-baseline-main/models/graspnet.py", line 33, in forward
    seed_features, seed_xyz, end_points = self.backbone(pointcloud, end_points)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/w/graspnet-baseline-main/models/backbone.py", line 105, in forward
    xyz, features, fps_inds = self.sa1(xyz, features)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1511, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/w/.local/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1520, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/w/graspnet-baseline-main/pointnet2/pointnet2_modules.py", line 235, in forward
    inds = pointnet2_utils.furthest_point_sample(xyz, self.npoint)
  File "/home/w/.local/lib/python3.8/site-packages/torch/autograd/function.py", line 553, in apply
    return super().apply(*args, **kwargs)  # type: ignore[misc]
  File "/home/w/graspnet-baseline-main/pointnet2/pointnet2_utils.py", line 71, in forward
    return _ext.furthest_point_sampling(xyz, npoint)
RuntimeError: Unknown layout

chenxi-wang, 2024-02-06:
Seems like an error in pointnet2. Did you compile pointnet2 correctly?

TheFive666, 2024-02-06:
yes
[image]

TheFive666, 2024-02-07:
> Seems like an error in pointnet2. Did you compile pointnet2 correctly?

yes,gcc version is 7.5.0

poplaryh, 2025-04-11:
is this problem solved? I encounter the same question as yours. Could you please give me some auggestions? really thx!