graspnet/graspnet-baseline #101: 对于6D grasp的一些疑问

quanfeifan, 2024-03-22:
6D grasp中（grasp.py）中，
1.rotation_matrix是相对于相机坐标系的吗？还是相对于别的坐标系？
2.translation是什么意思，是相对于哪个坐标系的呢？
3.已有rotation_matrix和translation，为什么还需要depth
4.如果需要计算世界坐标系下的夹取位姿，是不是先通过rotation_matrix和translation计算相机坐标系下的夹取位姿，再通过相机外参转换到世界坐标系呢？
问的有点多了，感谢大佬指导

quanfeifan, 2024-03-22:
看了一下rectGrasp类中的to_grasp函数，发现depth直接设置为0.02（graspAPI/grasp.py 744行左右），是不是意味着depth这一变量不太重要呢

pange1802703882, 2024-04-01:
请问您搞明白了吗，您是怎么转换的呀

chenxi-wang, 2024-04-01:
> 1.rotation_matrix是相对于相机坐标系的吗？还是相对于别的坐标系

相机坐标系

> 2.translation是什么意思，是相对于哪个坐标系的呢？

夹爪末端的位置，也是锚定的抓取点，相对于相机坐标系

> 3.已有rotation_matrix和translation，为什么还需要depth

因为translation是指对应的物体上的抓取点的坐标，夹爪到达这个位置以后还需要向前延伸一段距离才能把物体抓住，这个距离就是depth

> 4.如果需要计算世界坐标系下的夹取位姿，是不是先通过rotation_matrix和translation计算相机坐标系下的夹取位姿，再通过相机外参转换到世界坐标系呢？

是的，但要记得算位姿的时候把depth也加进去

> 看了一下rectGrasp类中的to_grasp函数，发现depth直接设置为0.02（graspAPI/grasp.py 744行左右），是不是意味着depth这一变量不太重要呢

2D平面抓取不好判断抓取深度，所以取定值简化，实际上depth的选择很影响成功率

quanfeifan, 2024-04-01:
> 相机坐标系
>
>
> 夹爪末端的位置，也是锚定的抓取点，相对于相机坐标系
>
>
> 因为translation是指对应的物体上的抓取点的坐标，夹爪到达这个位置以后还需要向前延伸一段距离才能把物体抓住，这个距离就是depth
>
>
> 是的，但要记得算位姿的时候把depth也加进去
>
>
> 2D平面抓取不好判断抓取深度，所以取定值简化，实际上depth的选择很影响成功率

感谢回复！

closed 2024-04-01 by quanfeifan