graspnet/graspnet-baseline #103: knn install error

matthewxialin, 2024-04-16:
4 years past. I cannot built correct enviroments(my env: pytorch 1.7.1, cuda 11). when i install knn, show err:

"C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.39.33519\bin\HostX86\x64\link.exe" /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:C:\Users\lin\miniconda3\envs\grasp\Lib\site-packages\torch\lib "/LIBPATH:C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\lib/x64" /LIBPATH:C:\Users\lin\miniconda3\envs\grasp\libs /LIBPATH:C:\Users\lin\miniconda3\envs\grasp /LIBPATH:C:\Users\lin\miniconda3\envs\grasp\PCbuild\amd64 "/LIBPATH:C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.39.33519\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.39.33519\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.22621.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\\lib\10.0.22621.0\\um\x64" c10.lib torch.lib torch_cpu.lib torch_python.lib cudart.lib c10_cuda.lib torch_cuda.lib /EXPORT:PyInit_knn_pytorch E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\cpu\knn_cpu.obj E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\cuda\knn.obj E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\vision.obj /OUT:build\lib.win-amd64-cpython-38\knn_pytorch\knn_pytorch.cp38-win_amd64.pyd /IMPLIB:E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\cpu\knn_pytorch.cp38-win_amd64.lib
  正在创建库 E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\cpu\knn_pytorch.cp38-win_amd64.lib 和对象 E:\graspnet\graspnet-baseline\knn\build\temp.win-amd64-cpython-38\Release\graspnet\graspnet-baseline\knn\src\cpu\knn_pytorch.cp38-win_amd64.exp
vision.obj : error LNK2001: unresolved external symbol "public: long * __cdecl at::Tensor::data_ptr<long>(void)const " (??$data_ptr@J@Tensor@at@@QEBAPEAJXZ)
vision.obj : error LNK2001: unresolved external symbol "struct THCState * state" (?state@@3PEAUTHCState@@EA)
build\lib.win-amd64-cpython-38\knn_pytorch\knn_pytorch.cp38-win_amd64.pyd : fatal error LNK1120: 2 unresolved external symbol
error: command 'C:\\Program Files\\Microsoft Visual

Vietisekai, 2024-06-10:
I get the same error
My env: python 3.9; pytorch 1.9; cu11.1

PINKCONCRETE, 2024-07-02:
same error

chenxi-wang, 2024-07-03:
Hi, the program was not tested on Windows. We recommend Linux.

Fang-Haoshu, 2025-02-17:
merged pull request, should be solved now

closed 2025-02-17 by Fang-Haoshu