graspnet/graspnet-baseline #105: About angles

dingthuang, 2024-05-02:
Hello, may I ask a question. When loading the dataset, labels for individual objects will be rotated. Only the view has been modified (use knn), but the angle has not been modified. This seems to cause issues with the label angle.

In short, rotation 1 is represented by view1 and angle1, while rotation 2 is represented by view2 and angle2. 
Rotate 1 through the rotation matrix to rotate 1 '(Rotate 2). 
View 1 rotates the rotation matrix to view 1 (view 2). (in your code knn)
Angle is not assign.
Rotation 1 ' (Rotate 2) cannot be represented by view1' and angle1.

TAO-TAO-TAO-TAO-TAO, 2024-05-09:
1：抱歉，我没有明白你的意思（And, in the training of inplane rotation classification, the cropped points are transformed using corresponding approach vector labels instead of predictions.）。 cropped points 是什么呢？
2：计算inplane 分类时候，你们的代码grasp_angle_class_loss = criterion_grasp_angle_class(grasp_angle_class_score, target_angles_cls)，其中target_angles_cls仅仅是来自于单个物体的标签，在转换到场景时候，并没有对其在旋转后重新计算inplane 正确的顺序。（我认为你们应该使用knn，重新计算一下inplane 的顺序）

dingthuang, 2024-05-09:
1：抱歉，我没有明白你的意思（And, in the training of inplane rotation classification, the cropped points are transformed using corresponding approach vector labels instead of predictions.）。 cropped points 是什么呢？
2：计算inplane 分类时候，你们的代码grasp_angle_class_loss = criterion_grasp_angle_class(grasp_angle_class_score, target_angles_cls)，其中target_angles_cls仅仅是来自于单个物体的标签，在转换到场景时候，并没有对其在旋转后重新计算inplane 正确的顺序。（我认为你们应该使用knn，重新计算一下inplane 的顺序）

chenxi-wang, 2024-05-11:
> 1：抱歉，我没有明白你的意思（And, in the training of inplane rotation classification, the cropped points are transformed using corresponding approach vector labels instead of predictions.）。 cropped points 是什么呢？

指的是 cinlinder grouping 中裁切的点云

> 2：计算inplane 分类时候，你们的代码grasp_angle_class_loss = criterion_grasp_angle_class(grasp_angle_class_score, target_angles_cls)，其中target_angles_cls仅仅是来自于单个物体的标签，在转换到场景时候，并没有对其在旋转后重新计算inplane 正确的顺序。（我认为你们应该使用knn，重新计算一下inplane 的顺序）

这是因为我们把坐标系转换到了单个物体用的坐标系，这样就避免了更繁琐的角度匹配