graspnet/graspnet-baseline #107: List of objects in seen ans unseen test set

1mingW, 2024-05-07:
Hi, 
Do you have a list of objects that are used in seen and unseen test sets? I retrained a model and want to do some experiment on seen and unseen objects, it would be great if there is some document that lists which object belongs to which test set.

chenxi-wang, 2024-05-11:
Hi, you can find the objects from object_ids.txt in each scene folder, and the object names can be found [here](https://graspnet.net/images/obj_list.pdf).

Fang-Haoshu, 2024-05-29:
@1mingW Hi, for the object list, you can also find the object list for both train and test using the graspnet_api. I remember there is a function that you can obtain all object ids given the input scene number. And the first 100 scenes are training set.

1mingW, 2024-06-02:
@Fang-Haoshu and @chenxi-wang Thank you for the answers!
I have another problem about how to collect the depth images. In real experiment, did you get point cloud directly or reconstructed with depth image? I was trying to reconstruct point clouds, but I got depth images with dtype uint16 by realsense pipeline and the depth value was not correct. I notice the depth images in your dataset have dtype int32. How did you save or collect the images?

1mingW, 2024-06-03:
I found I can actually reconstruct point clouds from depth images in uint16 format if deleting some values that are too large. But the quality of the point cloud is not as good as in your dataset: 
[image]
the RGB images have wrong color (which is a Realsense problem) and are shifted from the depth images. Did you also meet such problems while collecting data?
And I am still wondering if you use reconstructed point cloud or directly get point cloud from realsense topic (if you use ROS)?