graspnet/graspnet-baseline #117: How to make the model adopt the output of actual clamp size parameters？

xiaozhangniaibu, 2024-07-10:
I found that the depth and width of the gripper output by the model vary, but the actual gripper may not be able to achieve this opening level. How should I modify it so that the model output can be based on the actual gripper as a parameter?

In my understanding, it refers to the depth and width of the model output, which can be directly visualized, and all that can be modified are visualization parameters

xiaozhangniaibu, 2024-07-12:
I found that most of the output widths are between 0.1 and 0.05. If I manually filter out values greater than 0.07, it is very likely that there will be no grasping. How can I make the model predict to a fixed grasping size?
Moreover, due to the random sampling of the demo. py point cloud, the output results may vary each time. How should we handle this

FengMaverick, 2025-06-09:
> I found that the depth and width of the gripper output by the model vary, but the actual gripper may not be able to achieve this opening level. How should I modify it so that the model output can be [...]

Hello, I'd like to ask if you have solved this problem? I also want to try to reproduce GRASPNET with my parallel two-finger gripper. Now I can get a result, but I feel it doesn't apply to my gripper. Looking forward to your reply. Thank you!