graspnet/graspnet-baseline #127: data convention of camera_poses.npy

lif314, 2025-01-07:
Hi, thank you for your impressive work!
What does your camera space look like? 
```
OpenGL/MAYA       OpenCV/Colmap     Blender      Unity/DirectX     Unreal
Right-handed      Right-handed    Right-handed    Left-handed    Left-handed

     +y                +z           +z  +y         +y  +z          +z
     |                /             |  /           |  /             |
     |               /              | /            | /              |
     |______+x      /______+x       |/_____+x      |/_____+x        |______+x
    /               |                                              /
   /                |                                             /
  /                 |                                            /
 +z                 +y                                          +y
```

abhishek47kashyap, 2025-01-31:
See https://github.com/graspnet/graspnet-baseline/issues/93#issuecomment-1895368726

lif314, 2025-01-31:
@abhishek47kashyap Thank you!

closed 2025-01-31 by lif314