graspnet/graspnet-baseline #129: 部署到真机上，rotation_matrix对应不上

zhangbo9426426, 2025-02-21:
您好！非常感谢您的工作！
最近我在部署到我的机器人上进行应用时旋转矩阵上出现了一点问题一直无法解决

[image]
上面这幅图是我的机器人夹爪坐标系。按照我的夹爪坐标系和您api定义的夹爪坐标系应该是乘以一个变换矩阵可以应用，即
$$grasp.rotation\\_matrix*R_y(90)*R_z(180)$$
经过这个变换后的坐标系我发布在rviz中显示为上图中最下面的那个坐标系，正确的抓取姿态应该是下图圈住部分的夹爪姿态。结果还是不正确的，不知道是那里出现了问题

[image]
下面是我的部分坐标系转换代码
如果还需要更多信息可以随时补充，谢谢您！
`

    def coord_trans(T_obj_to_camera, robot_current_position):
     T_obj_to_camera = np.array(T_obj_to_camera)
     T_obj_to_camera = T_obj_to_camera.reshape(4, 4)
     T_obj_to_camera[:3, 3] *= 1000
     current_position = robot_current_position

     x_e, y_e, z_e, theta_x, theta_y, theta_z = current_position[:6]
     print("x_e, y_e, z_e, theta_x, theta_y, theta_z:", x_e, y_e, z_e, theta_x, theta_y, theta_z)
     # Get the transformation matrix from end effector to base
     T_end_effector_to_base = get_transformation_matrix(x_e, y_e, z_e, theta_x, theta_y, theta_z)

     # Compute the transformation from camera to base
     T_camera_to_base = np.dot(T_end_effector_to_base, T_camera_to_end_effector)
     T_object_to_base = np.dot(T_camera_to_base, T_obj_to_camera)
     translation = T_object_to_base[:3, 3]
     # 提取旋转矩阵
     rotation_matrix = T_object_to_base[:3, :3]
     print("rotation_matrix:", rotation_matrix)
     # 将旋转矩阵转换为欧拉角（假设顺序为XYZ）
     rotation = R.from_matrix(rotation_matrix)
     euler_angles = rotation.as_euler(seq, degrees=False)
     final_vector = np.concatenate([translation, euler_angles])
     print(final_vector)
     print("Translation Vector:", translation)
     print("Euler Angles (degrees):", euler_angles)
     return translation,euler_angles,final_vector

    def obj_to_camera_grasp(quaternion, position):
      """Compute the 4x4 homogeneous transformation matrix from quaternion and position."""
   
      rotation_matrix = quaternion_to_rotation_matrix(quaternion)
      transformation_matrix = np.eye(4) 
      transformation_matrix[:3, :3] = rotation_matrix 
      transformation_matrix[:3, 3] = position 

      return transformation_matrix

    if __name__ == "__main__":
      R_y_90 = np.array([[0, 0, 1],
                       [0, 1, 0],
                       [-1, 0, 0]])
      R_z_180 = np.array([[-1, 0, 0],
                        [0, -1, 0],
                        [0, 0, 1]])
      result = np.array([[ 0.02277477, -0.9989326 , -0.04018656],
                                  [ 0.49252543 , 0.04619145, -0.86907136],
                                  [ 0.87  ,      0.     ,     0.4930517 ]]) # 这里是graspnet检测出来的rotation_matrix
      result = np.dot(np.dot(result,R_y_90),R_z_180)  # 这里是右乘，左乘也试过结果也不对
      quaternion=rotation_matrix_to_quaternion(result)
      position = [-0.05231112 , 0.00762454 , 0.551]
      obj_to_camera_grasp = obj_to_camera_grasp(quaternion, position)
      current_position = [-550.91, -284.07, 546.11, 3.1415624272, 0, 0] # 这里是机器人末端到base_link的位姿，眼在手上
      translation,euler_angles,_=coord_trans(obj_to_camera_grasp, current_position)

`

zhangbo9426426, 2025-02-21:
或者我还试过下面这种方式来计算，在main中调用translation,euler_angles,_=coord_trans1(obj_to_camera_grasp, current_position)
结果仍然没有达到预期的姿态
```
def coord_trans1(T_obj_to_camera, robot_current_position):
    matrix_b = np.array([[0, 0, 1],    # 新的X轴对应原来的Z轴
                        [0, -1, 0],   # 新的Y轴对应原来的-Y轴
                        [1, 0, 0]])   # 新的Z轴对应原来的X轴
    T_obj_to_camera = np.array(T_obj_to_camera)
    T_obj_to_camera = T_obj_to_camera.reshape(4, 4)
    T_obj_to_camera[:3, 3] *= 1000
    current_position = robot_current_position

    x_e, y_e, z_e, theta_x, theta_y, theta_z = current_position[:6]
    print("x_e, y_e, z_e, theta_x, theta_y, theta_z:", x_e, y_e, z_e, theta_x, theta_y, theta_z)
    # Get the transformation matrix from end effector to base
    T_end_effector_to_base = get_transformation_matrix(x_e, y_e, z_e, theta_x, theta_y, theta_z)

    T_object_to_end_effector = np.dot(T_camera_to_end_effector, T_obj_to_camera)
    compensated_rotation_matrix = np.dot(T_object_to_end_effector[:3, :3],matrix_b)
    T_object_to_end_effector[:3, :3] = compensated_rotation_matrix
    T_object_to_base = np.dot(T_end_effector_to_base, T_object_to_end_effector)

    translation = T_object_to_base[:3, 3]
    # 提取旋转矩阵
    rotation_matrix = T_object_to_base[:3, :3]
    print("rotation_matrix:", rotation_matrix)
    # 将旋转矩阵转换为欧拉角（假设顺序为XYZ）
    rotation = R.from_matrix(rotation_matrix)
    euler_angles = rotation.as_euler(seq, degrees=False)
    final_vector = np.concatenate([translation, euler_angles])
    print(final_vector)
    print("Translation Vector:", translation)
    print("Euler Angles (degrees):", euler_angles)
    return translation,euler_angles,final_vector
```

zhangbo9426426, 2025-02-21:
[image]
上图物体附近处的坐标系对应的是下图中圈出来的姿态经过`result = np.dot(np.dot(result,R_y_90),R_z_180)`变换后的姿态，position验证过是正确的就没有每次修改验证。

[image]

closed 2025-02-22 by zhangbo9426426

LeeLeno, 2025-09-10:
所所以您现在解决了吗，我似乎感觉到像是坐标系没有对齐导致，但仍然还未改出正确版本。希望您能够慷慨赐教