graspnet/graspnet-baseline #134: After running "python setup.py install", "error: ‘c10::cuda::CUDACachingAllocator’ has not been declared" occurred, how should I solve it?

liangyijian1, 2025-04-15:
System: Ubuntu24.04(WSL2)
GPU:RTX A500 Laptop
CUDA: 10.2
Pytorch: 1.6.0

Below is the detailed error output:
```
(sim1) yijian@yiliang:~/repo/graspnet-baseline/knn$ python setup.py install
running install
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
writing knn_pytorch.egg-info/PKG-INFO
writing dependency_links to knn_pytorch.egg-info/dependency_links.txt
writing top-level names to knn_pytorch.egg-info/top_level.txt
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/utils/cpp_extension.py:335: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'knn_pytorch.egg-info/SOURCES.txt'
writing manifest file 'knn_pytorch.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_ext
building 'knn_pytorch.knn_pytorch' extension
g++ -pthread -B /home/yijian/miniconda3/envs/sim1/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DWITH_CUDA -I/home/yijian/repo/graspnet-baseline/knn/src -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/TH -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/yijian/miniconda3/envs/sim1/include/python3.8 -c /home/yijian/repo/graspnet-baseline/knn/src/cpu/knn_cpu.cpp -o build/temp.linux-x86_64-cpython-38/home/yijian/repo/graspnet-baseline/knn/src/cpu/knn_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=knn_pytorch -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/Parallel.h:149:0,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/all.h:7,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/extension.h:4,
                 from /home/yijian/repo/graspnet-baseline/knn/src/cpu/vision.h:2,
                 from /home/yijian/repo/graspnet-baseline/knn/src/cpu/knn_cpu.cpp:1:
/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/ParallelOpenMP.h:84:0: warning: ignoring #pragma omp parallel [-Wunknown-pragmas]
 #pragma omp parallel for if ((end - begin) >= grain_size)

/usr/local/cuda/bin/nvcc -DWITH_CUDA -I/home/yijian/repo/graspnet-baseline/knn/src -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/TH -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/yijian/miniconda3/envs/sim1/include/python3.8 -c /home/yijian/repo/graspnet-baseline/knn/src/cuda/knn.cu -o build/temp.linux-x86_64-cpython-38/home/yijian/repo/graspnet-baseline/knn/src/cuda/knn.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -DCUDA_HAS_FP16=1 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=knn_pytorch -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_75,code=sm_75 -std=c++14
/home/yijian/repo/graspnet-baseline/knn/src/cuda/knn.cu(231): warning: variable "t_k_16x16" was declared but never referenced

g++ -pthread -B /home/yijian/miniconda3/envs/sim1/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DWITH_CUDA -I/home/yijian/repo/graspnet-baseline/knn/src -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/TH -I/home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/yijian/miniconda3/envs/sim1/include/python3.8 -c /home/yijian/repo/graspnet-baseline/knn/src/vision.cpp -o build/temp.linux-x86_64-cpython-38/home/yijian/repo/graspnet-baseline/knn/src/vision.o -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=knn_pytorch -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/Parallel.h:149:0,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
[... 77 lines cut ...]
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/Context.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/ATen/ATen.h:5,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader_options.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader/base.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader/stateful.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data/dataloader.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/data.h:3,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/csrc/api/include/torch/all.h:4,
                 from /home/yijian/miniconda3/envs/sim1/lib/python3.8/site-packages/torch/include/torch/extension.h:4,
[... cut ...]
/home/yijian/repo/graspnet-baseline/knn/src/knn.h:33:42: error: ‘c10::cuda::CUDACachingAllocator’ has not been declared
     float *dist_dev = (float*)c10::cuda::CUDACachingAllocator::raw_alloc(ref_nb * query_nb * sizeof(float));
                                          ^~~~~~~~~~~~~~~~~~~~
/home/yijian/repo/graspnet-baseline/knn/src/knn.h:39:16: error: ‘c10::cuda::CUDACachingAllocator’ has not been declared
     c10::cuda::CUDACachingAllocator::raw_delete(dist_dev);
                ^~~~~~~~~~~~~~~~~~~~
error: command '/usr/bin/g++' failed with exit code 1
```

trltzy, 2025-05-28:
I have the same problem. Any solution?

fgc346, 2025-06-05:
> #include <c10/cuda/CUDACachingAllocator.h>
> 
> 添加到<knn/src/knn.h> 里面就行了

解决了我的knn的编译问题。

## 我的环境信息
```
OS : ubuntu 20.04
cuda: 11.6

pytorch : 1.8.0
cudatoolkit: 11.1
```